simbody/simbody #567: States cannot be used with different Systems

chrisdembia, 2017-06-21:
When we added support for parallel calculation of forces, we coupled State objects to the System they came from, by caching `Force*`s from the System in the State (see [here](https://github.com/simbody/simbody/blob/master/Simbody/src/GeneralForceSubsystem.cpp#L643) and [here](https://github.com/simbody/simbody/blob/master/Simbody/src/GeneralForceSubsystem.cpp#L713)).  This prevents users from easily using a state from one instance of a System with a different but identical instance of the System.

Related: https://github.com/opensim-org/opensim-core/issues/1455, https://github.com/opensim-org/opensim-core/issues/1785

sherm1, 2017-06-22:
Great find -- that's so wrong! Probably those pointers should have been ForceIndex values instead?

chrisdembia, 2017-06-22:
> Probably those pointers should have been ForceIndex values instead?

It'd be great if that will work.

PR #570 (2017-08-15): Do not store Force* in State cache.
Fixes #567.

This PR replaces the use of `Force*`s in the state cache with `ForceIndex`, as suggested by @sherm1 . I tested that these changes work fine with OpenSim, and that performance is not affected.

This PR will allow me to make a fix to `OpenSim::StatesTrajectory`, and may help @mitkof6 with some issues he had in OpenSim.
closed 2017-08-15 by sherm1