simbody/simbody #584: Replace pthread with c++11 threading support

chrisdembia, 2017-09-30:
Pthreads for windows has an LGPL license. In order to avoid that license, we should replace this library with C++11's std::thread, etc.

I've already started to work on this, and I am a little more than halfway through.

@sherm1 do you think you would you accept a PR (or PRs) for this?

Related: #422.

sherm1, 2017-09-30:
Definitely! I would love to get rid of pthreads altogether. It has other problems besides the license.

chrisdembia, 2017-10-01:
Great.

chrisdembia, 2017-10-02:
I've updated the top comment of this issue with a list of steps.

chrisdembia, 2017-12-15:
I have a branch, `replace_pthreads`, in which the removal of pthreads is nearly complete (the only thing left is to actually remove the pthreads files for Windows). I've tested fairly thoroughly on Mac, but not on other platforms. On Linux, I am getting the following error (note to self):

```
Visualizer listenerThread: unrecoverable error:
SimTK Exception thrown at VisualizerProtocol.cpp:179:
  Error detected by Simbody method VisualizerProtocol: An attempt to read() 2 bytes from pipe 5 failed with errno=9 (Bad file descriptor).
  (Required condition 'retval!=-1' was not met.)

```

My plan is to merge small parts of `replace_pthreads` into the `feature_replace_pthreads` branch until the removal is complete.

PR #605 (2018-01-08): Merge feature branch "feature_replace_pthreads" into master
Now that #602 is merged, the "replace_pthreads" feature is now complete. This PR simply merges `feature_replace_pthreads` into the master branch.

Fixes #584.

---
This change is [[image]](https://reviewable.io/reviews/simbody/simbody/605)
closed 2018-01-08 by sherm1