simbody/simbody #591: MobilizedBody::getMobilizerAcceleration() not implemented

apalazzi, 2017-11-21:
Hi,

the function MobilizedBody::getMobilizerAcceleration() is not implemented:

https://github.com/simbody/simbody/blob/master/Simbody/include/simbody/internal/MobilizedBody.h#L401

Bye
Andrea

apalazzi, 2017-11-28:
The code below shows the issue; the velocity and acceleration values are not exactly the same but they are quite near.

```
#include <cstdlib>
#include <iostream>
#include <Simbody.h>
using namespace SimTK;
class TranslationReporter : public PeriodicEventReporter {
public:
    TranslationReporter(const MultibodySystem& system,
    MobilizedBody::Translation& mobilizedBody,
    Real interval)
    :   PeriodicEventReporter(interval), system(system), _mobilizedBody(mobilizedBody) {}
    void handleEvent(const State& s) const override {
        std::cout << "velocity: " << _mobilizedBody.getMobilizerVelocity(s).norm()
        		<< " accel: " << _mobilizedBody.getMobilizerAcceleration(s).norm()
				<< std::endl;
        }
private:
    const MultibodySystem& system;
    MobilizedBody::Translation& _mobilizedBody;
};
int main() {
    // Define the system.
    MultibodySystem system;
    SimbodyMatterSubsystem matter(system);
    GeneralForceSubsystem forces(system);
    Force::UniformGravity gravity(forces, matter, Vec3(0.0, -9.81, 0.0));
    Body::Rigid body(MassProperties(10.0, Vec3(0, 0, 0), Inertia(0.1)));
    body.addDecoration(Transform(), DecorativeSphere(0.1));
    MobilizedBody::Translation carriage(matter.Ground(), body);
    Constraint::PrescribedMotion(matter,
    		new Function::Sinusoid(1.0, 1.0, 0.0),
//			new Function::Constant(0.0),
    		carriage, MobilizerQIndex(0));
    Constraint::PrescribedMotion(matter,
    		new Function::Constant(10.0),
			carriage, MobilizerQIndex(1));
    Constraint::PrescribedMotion(matter,
    		new Function::Constant(0.0),
			carriage, MobilizerQIndex(2));
    system.addEventReporter(new TranslationReporter(system, carriage, 0.1));
    // Set up visualization.
    Visualizer viz(system);
    viz.setShowFrameRate(true);
    viz.setShowSimTime(true);
    viz.setSystemUpDirection(YAxis);
    system.addEventReporter(new Visualizer::Reporter(viz, 1.0/30.0));
    // Initialize the system and state.
    State state = system.realizeTopology();
    RungeKuttaMersonIntegrator integ(system);
    TimeStepper ts(system, integ);
    ts.initialize(state);
    ts.stepTo(20.0);
    return 0;
}
```

sherm1, 2018-01-08:
The problem is that MobilizedBody::getMobilizerAcceleration() is an unimplemented method. Here is the doxygen comment and code for it from MobilizedBody.h:
```c++
/** TODO: Not implemented yet -- any volunteers? At stage Acceleration, return
the cross-mobilizer acceleration A_FM, the relative acceleration of body B's 
"moving" mobilizer frame M in the parent body's corresponding "fixed" frame F, 
measured and expressed in F. Note that this isn't the usual spatial acceleration 
since it isn't expressed in G. **/
const SpatialVec& getMobilizerAcceleration(const State& state) const { // A_FM
    SimTK_ASSERT_ALWAYS(!"unimplemented method", 
"MobilizedBody::getMobilizerAcceleration() not yet implemented -- volunteers?");
    return *(new SpatialVec());
}
```

sherm1, 2018-01-08:
Hi, Andrea. The problem here is that you are not getting error messages from Simbody because
- there is no try/catch block around your main program (required for all Simbody programs)
- you need to make a Debug build in order to get all error checking in place (many of those checks are expensive so are suppressed in Release builds).

So your main problem should look like this:
```c++
int main() {
  try {
    // Define the system.
    MultibodySystem system;
    // ...
    ts.stepTo(20.0);
  } catch (const std::exception& e) {
     std::cout << "Exception: " << e.what() << std::endl;
  }
  return 0;
```
I added that to your program and ran it in a Debug build and got:
```
velocity: 1 accel: 1.42109e-15
Exception: SimTK Exception thrown at StateImpl.h:1460:
  Expected stage to be at least Acceleration in StateImpl::getUDot(subsys) but current stage was Velocity
```
That message means you must "realize" accelerations in the State in the handleEvent() method before you can ask for them. Change handleEvent() like this:
```c++
    void handleEvent(const State& s) const override {
        system.realize(s, Stage::Acceleration);
        std::cout << "velocity: " << _mobilizedBody.getMobilizerVelocity(s).norm()
        		<< " accel: " << _mobilizedBody.getMobilizerAcceleration(s).norm()
				<< std::endl;
    }
```
After I made that change I reran the program (still in Debug so all error checking is on). Here are the first few lines of output:
```
velocity: 1 accel: 1.42109e-15
velocity: 0.995001 accel: 0.0998334
velocity: 0.980057 accel: 0.198669
velocity: 0.955336 accel: 0.29552
velocity: 0.920893 accel: 0.389418
velocity: 0.877583 accel: 0.479426
velocity: 0.825336 accel: 0.564642
velocity: 0.764842 accel: 0.644218
velocity: 0.696707 accel: 0.717356
```
As you can see the accelerations do not look at all like the velocities!

Regards,
Sherm

closed 2018-01-08 by sherm1