simbody/simbody #593: Build fails with intel compiler. Error: more than one instance of overloaded function matches the argument list

CSPer, 2017-12-01:
The error message is as follows:

      simbody-Simbody-3.5.4/SimTKcommon/SmallMatrix/include/SimTKcommon/internal/Vec.h(69): error: more than one instance of overloaded function "SimTK::Impl::conformingSubtract" matches the argument list:

      function template "void SimTK::Impl::conformingSubtract(const SimTK::Vec<1, E1, S1> &, const SimTK::Vec<1, E2, S2> &, SimTK::Vec<1, SimTK::CNT<E1>::Result<E2>::Sub, 1> &)"

      function template "void SimTK::Impl::conformingSubtract(const SimTK::Vec<N, E1, S1> &, const SimTK::Vec<N, E2, S2> &, SimTK::Vec<N, SimTK::CNT<E1>::Result<E2>::Sub, 1> &)"
 
      argument types are: (const SimTK::Vec<1, double, 3>, const SimTK::Vec<1, SimTK::Real, 1>, SimTK::Vec<1, SimTK::Real, 1>)

      conformingSubtract(reinterpret_cast<const Vec<N-1,E1,S1>&>(r1), 
      ^
          detected during:
            instantiation of "void SimTK::Impl::conformingSubtract(const SimTK::Vec<N, E1, S1> &, const SimTK::Vec<N, E2, S2> &, SimTK::Vec<N, SimTK::CNT<E1>::Result<E2>::Sub, 1> &) [with N=2, E1=double, S1=3, E2=SimTK::Real={SimTK_Real={double}}, S2=1]" at line 72
            instantiation of "void SimTK::Impl::conformingSubtract(const SimTK::Vec<N, E1, S1> &, const SimTK::Vec<N, E2, S2> &, SimTK::Vec<N, SimTK::CNT<E1>::Result<E2>::Sub, 1> &) [with N=3, E1=double, S1=3, E2=SimTK::Real={SimTK_Real={double}}, S2=1]" at line 563
            instantiation of "SimTK::Vec<M, SimTK::CNT<ELT>::Result<EE>::Sub, 1> SimTK::Vec<M, ELT, STRIDE>::conformingSubtract(const SimTK::Vec<M, EE, SS> &) const [with M=3, ELT=double, STRIDE=3, EE=SimTK::Real={SimTK_Real={double}}, SS=1]" at line 413 of "simbody-Simbody-3.5.4/SimTKcommon/SmallMatrix/include/SimTKcommon/internal/ResultType.h"
            instantiation of "SimTK::SubCNTs<NRow, 1, LArgDepth, SimTK::Vec<NRow, LE, LRS>, LCS, LRS, NRow, 1, RArgDepth, SimTK::Vec<NRow, RE, RRS>, RCS, RRS>::Type SimTK::SubCNTs<NRow, 1, LArgDepth, SimTK::Vec<NRow, LE, LRS>, LCS, LRS, NRow, 1, RArgDepth, SimTK::Vec<NRow, RE, RRS>, RCS, RRS>::perform(const SimTK::SubCNTs<NRow, 1, LArgDepth, SimTK::Vec<NRow, LE, LRS>, LCS, LRS, NRow, 1, RArgDepth, SimTK::Vec<NRow, RE, RRS>, RCS, RRS>::L &, const SimTK::SubCNTs<NRow, 1, LArgDepth, SimTK::Vec<NRow, LE, LRS>, LCS, LRS, NRow, 1, RArgDepth, SimTK::Vec<NRow, RE, RRS>, RCS, RRS>::R &) [with NRow=3, LArgDepth=1, LE=double, LCS=9, LRS=3, RArgDepth=1, RE=SimTK::Real={SimTK_Real={double}}, RCS=3, RRS=1]" at line 1023
            instantiation of "SimTK::Vec<M, E1, S1>::Result<SimTK::Vec<M, E2, S2>>::Sub SimTK::operator-(const SimTK::Vec<M, E1, S1> &, const SimTK::Vec<M, E2, S2> &) [with M=3, E1=double, S1=3, E2=SimTK::Real={SimTK_Real={double}}, S2=1]" at line 596 of "simbody-Simbody-3.5.4/SimTKcommon/SmallMatrix/include/SimTKcommon/internal/Mat.h"
            instantiation of "SimTK::Mat<M, N, ELT, CS, RS>::Result<SimTK::Mat<M, N, E2, CS2, RS2>>::Sub SimTK::Mat<M, N, ELT, CS, RS>::conformingSubtract(const SimTK::Mat<M, N, E2, CS2, RS2> &) const [with M=3, N=3, ELT=double, CS=1, RS=3, E2=SimTK::Real={SimTK_Real={double}}, CS2=3, RS2=1]" at line 518 of "simbody-Simbody-3.5.4/SimTKcommon/SmallMatrix/include/SimTKcommon/internal/ResultType.h"
            instantiation of "SimTK::SubCNTs<NRow, NCol, LArgDepth, SimTK::Mat<NRow, NCol, LE, LCS, LRS>, LCS, LRS, NRow, NCol, RArgDepth, SimTK::Mat<NRow, NCol, RE, RCS, RRS>, RCS, RRS>::Type SimTK::SubCNTs<NRow, NCol, LArgDepth, SimTK::Mat<NRow, NCol, LE, LCS, LRS>, LCS, LRS, NRow, NCol, RArgDepth, SimTK::Mat<NRow, NCol, RE, RCS, RRS>, RCS, RRS>::perform(const SimTK::SubCNTs<NRow, NCol, LArgDepth, SimTK::Mat<NRow, NCol, LE, LCS, LRS>, LCS, LRS, NRow, NCol, RArgDepth, SimTK::Mat<NRow, NCol, RE, RCS, RRS>, RCS, RRS>::L &, const SimTK::SubCNTs<NRow, NCol, LArgDepth, SimTK::Mat<NRow, NCol, LE, LCS, LRS>, LCS, LRS, NRow, NCol, RArgDepth, SimTK::Mat<NRow, NCol, RE, RCS, RRS>, RCS, RRS>::R &) [with NRow=3, NCol=3, LArgDepth=1, LE=double, LCS=1, LRS=3, RArgDepth=1, RE=SimTK::Real={SimTK_Real={double}}, RCS=3, RRS=1]" at line 1256 of "simbody-Simbody-3.5.4/SimTKcommon/SmallMatrix/include/SimTKcommon/internal/Mat.h"
            instantiation of "SimTK::Mat<M, N, EL, CSL, RSL>::Result<SimTK::Mat<M, N, ER, CSR, RSR>>::Sub SimTK::operator-(const SimTK::Mat<M, N, EL, CSL, RSL> &, const SimTK::Mat<M, N, ER, CSR, RSR> &) [with M=3, N=3, EL=double, CSL=1, RSL=3, ER=SimTK::Real={SimTK_Real={double}}, CSR=3, RSR=1]" at line 760 of "simbody-Simbody-3.5.4/SimTKcommon/Mechanics/include/SimTKcommon/internal/SpatialAlgebra.h"

When compiling via gcc, the error disappears and it builds successfully

ericfonseca95, 2023-05-29:
Hello, 

I receive a similar error when attempting to compile simbody on a computing cluster with 
intel/2018.1.163
openmpi/3.1.0

Compiling with gcc also eliminates the error, but it would be great to be able to use the intel compiler. 

Eric

sherm1, 2023-05-29:
We would be delighted to accept a PR that permits Simbody to build with the Intel compiler (without breaking the other ones!). Anyone with Intel access want to submit a fix?