simbody/simbody #604: MobilizedBody::getMobilizerAcceleration() should be virtual

apalazzi, 2018-01-04:
Hi,

the method MobilizedBody::getMobilizerAcceleration() is overriden in MobilizedBody::Translation but is not declared virtual; if a Translation class is then created with new and its pointer stored in a MobilizedBody* pointer, the call to getMobilizerAcceleration ends up executing MobilizedBody::getMobilizerAcceleration() and not Translation::getMobilizerAcceleration().

That is:
```
MobilizedBody* mob=new MobilizedBody::Translation();
mob->getMobilizerAcceleration(); // calls MobilizedBody::getMobilizerAcceleration
```

If the method is supposed to be overriden in subclasses it should be declared as virtual; or, if the Translation::getMobilizerAcceleration method is generic enough it should be moved to MobilizedBody.

sherm1, 2018-01-08:
None of the MobilizedBody interface methods are supposed to be virtual. The virtual methods are located in the MobilizedBodyImpl class and its derived classes. There are two problems here:
1. The `MobilizedBody::getMobilizerAcceleration()` method is not implemented (it does say that clearly in its documentation but admittedly that's not very helpful!), and
2. The MobilizedBody::Translation class (like all the other joints) has its own more-specific API. In this case that joint implements a method it calls `getMobilizerAcceleration()` which uses the same name as the unimplemented MobilizedBody method. It should have been called something more specific like `getTranslationAcceleration()` so it wouldn't interfere. You can see it isn't actually the same method at all since it has a different return type (Vec3 vs. SpatialVec).

The Translation mobilizer is the only Simbody joint whose API has a method `getMobilizerAcceleration()`; that is really an API bug and shouldn't be there. All it does is call getUDot(), which is supported by every joint type. Or you can access udots generically with [`MobilizedBody::getUDotAsVector()`](https://simbody.github.io/simbody-master-doxygen/api/classSimTK_1_1MobilizedBody.html#af14203212503b70930c104cf35aa45b1).

Can you use getUDot() or getUDotAsVector() and forget about using getMobilizerAcceleration() for now? They are not exactly the same thing for some joints because udots are the time derivatives of a mobilizer's velocity coordinates u. For a translation joint the u's are the individual components of the translation vector from the mobilizer's F origin on the parent body to the M origin on the child (that is, V_FM) so the time derivative udot=A_FM, the acceleration of M's origin in F, which is what MobilizedBody::getMobilizerAcceleration() will return when it is implemented. But the mapping from udo t to A_FM can be much more complicated for some joints which is presumably why the general getMobilzierAcceleration() method hasn't been implemented yet.

Sherm

sherm1, 2018-01-08:
I retitled this issue "MobilizedBody::Translation has confusing getMobilizerAcceleration() method"

sherm1, 2018-01-08:
Andrea -- I addressed this by deprecating the badly-named methods in MobilizedBody::Translation and replacing them with names that don't conflict with the base class. The problem still exists that the base class method MobilizedBody::getMobilizerAcceleration() is unimplemented, but hopefully this will reduce future confusion.

Sorry for the trouble, and thanks for reporting it.

closed 2018-01-08 by sherm1

apalazzi, 2018-01-08:
Hi,

thank you for the support and reactivity, and I'm glad I can somehow contribute to this project :)

sherm1, 2018-01-08:
@apalazzi, your contributions as a user are much appreciated!  FYI, we also wlecome code and documentation contributions from the community in case you don't like waiting around for fixes to get done. Please see [CONTRIBUTING.md](https://github.com/simbody/simbody/blob/master/CONTRIBUTING.md) for more info.

PR #608 (2018-01-08): Deprecates some badly-named methods in MobilizedBody::Translation
The Translation mobilizer had some methods whose names were confusingly similar to methods in the MobilizedBody base class, causing unnecessary confusion. This PR changes those names and deprecates the bad ones.

I also had to re-enable the deprecation warning C4996 in Visual Studio. That had been disabled globally meaning the DEPRECATION_14 macro was non-functional. I switched to locally disabling the warning in the one .cpp file that seemed to be instantiating the troublesome code. I *think* that avoids the problem altogether  [... cut]
commit 8dd957f5 (2018-01-08): Deprecates some badly-named methods in MobilizedBody::Translation (#608)

Deprecates some badly-named methods in MobilizedBody::Translation.
Re-enables the deprecation warning C4996 in Visual Studio.
Fixes #604. Fixes #607.