simbody/simbody #606: Release Simbody 3.6

chrisdembia, 2018-01-07:
As mentioned in #510, it would be nice to get a new release of Simbody into Ubuntu 18.04. As discussed with @sherm1, I will start preparing this release after #605 is merged. The version will be 3.6.

If OpenSim 4.0 requires additional changes to Simbody, we can create an additional release.

To create the release, I will follow the TODO list in #275.

cc @aseth1 @aymanhab @j-rivero @scpeters @traversaro

traversaro, 2018-01-08:
Thanks @chrisdembia . Let me know if we can help in any way as "external" users.

traversaro, 2018-01-08:
I went through the README and one thing I noticed is that the recommended/tested version of Visual Studio (2015) is not easily downloadable any more from  https://www.visualstudio.com .

chrisdembia, 2018-01-08:
Thanks @traversaro. I've attempted to update the README in this branch: https://github.com/simbody/simbody/tree/simbody-3.6-prep.

I added Visual Studio 2017 to the README, and based on your comment, I also added a link for obtaining the older 2015 version. OpenSim still uses Visual Studio 2015, so I do not want to remove 2015 from the README.

Even though AppVeyor uses 2015, we have ensured locally that Simbody works with 2017.

scpeters, 2018-01-08:
Appveyor does [support VS 2017](https://www.appveyor.com/updates/) now, though I understand if you don't want to test with both on each commit, since it can be slow.

chrisdembia, 2018-01-08:
Thanks @scpeters. Our Appveyor account is shared between multiple projects and we do not want to add more jobs to the queue. My opinion is that we should only move if we depend on features in 2017. Otherwise, it's nice to ensure that we still support using the older compiler.

chrisdembia, 2018-01-08:
@j-rivero should I change anything in the debian directory, or should I let you handle those changes? Is there anything I can do to help address #545?

j-rivero, 2018-01-08:
> @j-rivero should I change anything in the debian directory

some minimal changes pull/610

> Is there anything I can do to help address #545?

To solve the problem in already released Ubuntu distros, we would need to fill a [SRU (Stable Release Update)](https://wiki.ubuntu.com/StableReleaseUpdates) which requires to fix a major problem with the package, which in this case is kind of subjective but could work. We can try it, I just need some time to look into the problem, fix it and create all the SRU information.

chrisdembia, 2018-01-08:
Thank you! Oh I meant for the upcoming 3.6 package; I don't think it is necessary to repair the existing package.

scpeters, 2018-01-09:
I did some brief testing of gazebo with simbody 3.6 today, and I didn't see any problems.

chrisdembia, 2018-01-09:
Whew :)

j-rivero, 2018-01-11:
> Thank you! Oh I meant for the upcoming 3.6 package; I don't think it is necessary to repair the existing package.

That should not be difficult. We probably will need another package that ships the binaries (`simbody` could be use) since all that we have now are the libraries.

chrisdembia, 2018-01-13:
@sherm1, here is doxygen to post for 3.6: https://www.dropbox.com/s/iltgte4pk7x76mt/simbody-3.6-doxygen.zip?dl=0

chrisdembia, 2018-01-13:
If there is a way I can help with posting the doxygen on simtk, let me know.

chrisdembia, 2018-01-14:
@sherm1 after #612, #610, and #611 are merged, I can create the `simbody-3.6` branch and `Simbody-3.6` tag.

Here is proposed text for the [releases](https://github.com/simbody/simbody/releases) page and the [annotated tag](https://git-scm.com/book/en/v2/Git-Basics-Tagging) message:

This is the first release of Simbody that requires C++11, as much of the code base now takes advantage of C++11 (e.g., `SimTK::Array_`, `SimTK::ResetOnCopy`, `SimTK::ReinitOnCopy`, and move constructor for `SimTK::State`). The MinGW compiler is now supported, providing an alternative to Visual C++ on Windows. The state's caching mechanism is now more fine-grained  ("stage versions"), though this change should not affect users in this release.

If upgrading from 3.5, you may need to make minor changes to downstream code:
  - `SimTK::ClonePtr`'s equality comparison operator previously checked equality between the underlying objects, but now checks equality between pointers.
  - Support for `long double` has been removed.
  - Some methods in `SimTK::ClonePtr` and `SimTK::ReferencePtr` have been renamed.
  - `SimTK::Xml` was changed from a class to a namespace.

Where possible, we have deprecated previous methods rather than removing them completely.

There were numerous smaller improvements to Simbody since the previous release, in build and installation, documentation, performance, bug fixes, and small enhancements. For details, see [CHANGELOG.md](https://github.com/simbody/simbody/blob/Simbody-3.6/CHANGELOG.md).

For installation instructions, see [README.md](https://github.com/simbody/simbody/blob/Simbody-3.6/README.md).
For information on contributing to the Simbody project, see [CONTRIBUTING.md](https://github.com/simbody/simbody/blob/master/CONTRIBUTING.md).

If you have questions or problems, please post to the [Simbody Forum](https://simtk.org/forums/viewforum.php?f=47).

sherm1, 2018-01-14:
@chrisdembia, I posted the doxygen in new repo simbody/simbody-3.6-doxygen. It is viewable [here](https://simbody.github.io/simbody-3.6-doxygen/api/index.html). You are an admin on that repo in case you want to make any changes.

sherm1, 2018-01-14:
@chrisdembia, re the release text -- looks great. I edited to remove the C++11 enumeration -- enough just to say Simbody depends on C++11 features.

chrisdembia, 2018-01-14:
> @chrisdembia, I posted the doxygen in new repo simbody/simbody-3.6-doxygen.

Oh right I forgot we moved the hosting of the doxygen. The website looks good to me

chrisdembia, 2018-01-14:
> @chrisdembia, re the release text -- looks great. I edited to remove the C++11 enumeration -- enough just to say Simbody depends on C++11 features.

Thanks for reviewing this.

chrisdembia, 2018-01-14:
@sherm1 There are some places where we want an unchanging link to the latest release's doxygen. Previously, we had https://simtk.org/api_docs/simbody/latest/index.html. To preserve this going forward, I created a new repo `simbody-latest-doxygen` whose `index.html` we can easily edit to point to the doxygen for the latest release, via https://simbody.github.io/simbody-latest-doxygen.

Would it be possible to edit https://simtk.org/api_docs/simbody/latest/index.html to just forward to https://simbody.github.io/simbody-latest-doxygen ?

sherm1, 2018-01-14:
> Would it be possible to edit https://simtk.org/api_docs/simbody/latest/index.html to just forward to https://simbody.github.io/simbody-latest-doxygen ?

Great idea. Yes, but I don't think I have the ability to edit simtk.org files any more. Can you ask Henry to edit it?

chrisdembia, 2018-01-14:
Okay; I've asked Henry.

marnunez, 2018-02-12:
Hi! I think this stopped mid process: there's still no 3.6 tag yet the master opensim-core asks for it. Is that so?

chrisdembia, 2018-02-12:
Hey @marnunez Since the 3.5 release, Simbody's master branch has been labeled as "3.6" in CMake. OpenSim's `find_package()` line has required Simbody "3.6" for over a year, but that does not mean OpenSim is looking for a 3.6 tag.

closed 2018-02-21 by chrisdembia