simbody/simbody #607: Method deprecation not working on Windows any more

sherm1, 2018-01-08:
Simbody's `DEPRECATED_14()` macro is not functioning on Windows any more. I believe this is because warning 4996 has been disabled globally. That should only be disabled locally with pragma push/pop so that we can mark methods deprecated.

@chrisdembia do you remember why 4996 got removed globally?

chrisdembia, 2018-01-08:
Uh oh. This warning got removed when we replaced `memcpy` with `std::copy` this summer (https://github.com/simbody/simbody/blob/a899db7be06ff3ae70c4b60a8922e62b5b6a8a63/CMakeLists.txt#L456). I think @antoinefalisse tried disabling the warning in a more local manner but MSVC wouldn't let us.

See here for more information: https://stackoverflow.com/questions/42003997/c-array-copy-showing-errors-in-vc

sherm1, 2018-01-08:
OK, thanks. I'll see if I can deal with it locally.

PR #608 (2018-01-08): Deprecates some badly-named methods in MobilizedBody::Translation
The Translation mobilizer had some methods whose names were confusingly similar to methods in the MobilizedBody base class, causing unnecessary confusion. This PR changes those names and deprecates the bad ones.

I also had to re-enable the deprecation warning C4996 in Visual Studio. That had been disabled globally meaning the DEPRECATION_14 macro was non-functional. I switched to locally disabling the warning in the one .cpp file that seemed to be instantiating the troublesome code. I *think* that avoids the problem altogether  [... cut]
commit 8dd957f5 (2018-01-08): Deprecates some badly-named methods in MobilizedBody::Translation (#608)

Deprecates some badly-named methods in MobilizedBody::Translation.
Re-enables the deprecation warning C4996 in Visual Studio.
Fixes #604. Fixes #607.
closed 2018-01-08 by sherm1