simbody/simbody #628: Build simbody with vcpkg on windows

Fleischner, 2018-03-21:
https://github.com/Microsoft/vcpkg is microsofts intended way to build libraries from source on windows.

Bring simbody to it too. All required deps are there:
openblas, lapack, opengl, freeglut

Attached are the portfiles samples and the patched main CMakeLists:
- Cmakelists changed at BUILD_USING_OTHER_LAPACK as I had problems to pass a Path with ";" as cmake command option.
- CMakeLists should also remove overrides for CXX_FLAGS. These get passed in by vcpkg
- Portfiles not completly finished. I still get some failures on tests. (Maybe need to patch the lapack headers with the headers from the vcpkg lapack port)
- Visualization disabled. Build problems, as the sources have lot of custom patches for windows. Should just use freeglut too.

- Files renamed with .txt for upload

[CMakeLists.txt](https://github.com/simbody/simbody/files/1835005/CMakeLists.txt)
[CONTROL.txt](https://github.com/simbody/simbody/files/1835006/CONTROL.txt) 
[portfile.txt](https://github.com/simbody/simbody/files/1835004/portfile.txt)

chrisdembia, 2018-03-21:
Hey @Fleischner it's awesome that you explored getting Simbody into vcpkg. Would you be interested in submitting a PR with your changes?

traversaro, 2018-03-21:
Related vcpkg issue: https://github.com/Microsoft/vcpkg/issues/2955 .

chrisdembia, 2018-03-21:
Oh, neat; thanks @traversaro.

chrisdembia, 2018-03-22:
Thanks @Fleischner. I don't have time to work on this in the near future, but having those files available will really help.

Ace314159, 2021-10-15:
If #724 is merged, if the `WINDOWS_USE_EXTERNAL_LIBS` CMake variable is set, I think that would solve the that problem as well.