simbody/simbody #636: Gazebo2Simbody can't generate correct simbody model for gazebo static model

jacknlliu, 2018-05-26:
From [gazebo issue 2466](https://bitbucket.org/osrf/gazebo/issues/2466/all-static-models-are-set-to-original), we find that the static model can't be set on correct pose in the gazebo world when using simbody physics engine.

We add two static models 'table', but they are all set on the origin point. But actually they should like

[image]

But using [examples/Gazebo2Simbody](https://github.com/simbody/simbody/blob/master/examples/Gazebo2Simbody/Gazebo2Simbody.cpp) or gazebo with simbody, we get something like

[image] 

The link frame seems right, but the visual parts of two static models are on the origin point.

The SDF file is [here](https://github.com/jacknlliu/development-issues/blob/master/assets/tables_test.sdf).

sherm1, 2018-05-29:
Thanks for reporting this, @jacknlliu. I looked at the Gazebo2Simbody example (which has a crude partial sdf parser built in). I found this at [Gazebo2Simbody.cpp:627](https://github.com/simbody/simbody/blob/master/examples/Gazebo2Simbody/Gazebo2Simbody.cpp#L627):
```c++
if (model.isStatic) {
    mobod = matter.updGround();
} else if ...
```
indicating that I thought static meant "attached to the ground (world) body". The tables are actually attached to their own links though, with those links fixed to world instead. Although I don't think this is the right fix, commenting that out to read:
```c++
/*if (model.isStatic) {
    mobod = matter.updGround();
} else*/ if ...
```
produced the right starting image:
[image]

I didn't look in Gazebo but I wouldn't be surprised if it also has the same mistake, probably copied over from this example.

There is still a problem, at least in Gazebo2Simbody. It still doesn't understand that the links in the static models should be _welded_ to world. Instead it is making them _free_ with respect to world. If you allow the simulation to run you can see the tables bounce around (tighten the Accuracy setting from 0.1 to 0.01 to get a stable simulation where they sit still).

I _think_ the best fix would be that the MultibodyGraphMaker should be told that the links are static and that it should create weld joints for them rather than free joints, when no other joint is specified.

A workaround in the sdf file, with the same effect, would be to make the table models non-static, but specify weld joints for the links to make them effectively static.

jacknlliu, 2018-05-29:
@sherm1 Thanks for your reply. I find that the codes of Gazebo are copied over from this example, and [here](https://bitbucket.org/osrf/gazebo/src/f34784b7994ade9758440d7f65bd6c4e295dcc9e/gazebo/physics/simbody/SimbodyPhysics.cc#lines-923) has the same mistake.

I think this example is the best reference for gazebo, hope it will be fixed by simbody community. Then we can cast it to gazebo.

jacknlliu, 2018-06-24:
I try to fix it. But which one is the appropriate method for this, `Constraint::Weld` or `MobilizedBody::Weld`? I'm not familiar with the simbody terminology and API. Any more help about this? @sherm1

sherm1, 2018-06-24:
Hi, Jack. MobilizedBody::Weld would be the right choice here. That is _much_ more efficient (essentially free since there are zero dofs) than Constraint::Weld which would add 6 constraint equations to prevent motion.

In internal coordinate multibody systems, tree joints (what Simbody calls "mobilizers") are implemented with one differential equation per degree of freedom, so for example one equation for a pin (revolute) joint or sliding (prismatic) joint, but six equations for a free joint. Constraints are the opposite -- it would take 5 constraint equations to implement a pin joint, and zero for a free "constraint".

sherm1, 2018-06-24:
Basically, for bodies that have no connection to the world we need to distinguish the ones that should be free to move from those that should be welded to ground. Currently MultibodyGraphMaker adds a Free mobilizer (six dofs) to those unconnected bodies; for static bodies it should b adding a Weld mobilizer (zero dofs) instead.