simbody/simbody #637: Analysis of EFF convergence algorithm

Gjacquenot, 2018-06-18:
I am using Simbody contact algorithm and I would like advice on the convergence of the program below. I am evaluating the contact forces between two boxes, and as I increase resolution of boxes geometries, result diverge. Result is exact with lowest resolution, but diverge to another value 20% higher. Maybe it is totally normal, however I would like experts advice (@sherm1 , @chrisdembia ;-))

Geometries involved are a plane `ContactGeometry::HalfSpace()` and
a cube `PolygonalMesh::createBrickMesh`

Analytical solution is `stiffness * penetration * area_of_contact`, where `stiffness` is 100, `penetration` is 0.05

Numerical solution can thus be compared to the analytical one.

A cube of length 0.5 is created with `createBrickMesh(Vec3(0.25,0.25,0.25), i)`
where is the resolution. `area_of_contact` is thus `0.5*0.5`

Resolution is progressively increased from 1 to 30. Results are exact for
resolution=1 and 2, but increased with resolution and seems to stabilize
around 20% higher.

Penetration is set to 0.05 and center of the cube is placed at (0,0.25-0.25,0.0)

Force is along Y axis.

Convergence test were performed on similar case with sphere
https://github.com/simbody/simbody/blob/master/Simbody/tests/TestElasticFoundationForce.cpp

Here is the program

    #include "SimTKsimbody.h"

    using namespace SimTK;
    using namespace std;

    const Real TOL = 1e-5;

    #define ASSERT(cond) {SimTK_ASSERT_ALWAYS(cond, "Assertion failed");}

    template <class T>
    void assertEqual(T val1, T val2) {
        ASSERT(abs(val1-val2) < TOL || abs(val1-val2)/max(abs(val1), abs(val2)) < TOL);
    }

    template <int N>
    void assertEqual(Vec<N> val1, Vec<N> val2) {
        for (int i = 0; i < N; ++i)
            assertEqual(val1[i], val2[i]);
    }

    /**
     * Program to test convergence of EFF algorithm from collision of a box on a
     * plane. Geometries involved are a plane `ContactGeometry::HalfSpace()` and
     * a cube `PolygonalMesh::createBrickMesh`
     * 
     * Analytical solution is stiffness * penetration * area_of_contact
     *
     * Numerical solution can thus be compared to the analytical one.
     *
     * A cube of length 0.5 is created with `createBrickMesh(Vec3(0.25,0.25,0.25), i)`
     * where is the resolution.
     *
     * Resolution is progressively increased from 1 to 60. Result are exact for
     * resolution=1 and 2, but increased with resolution and seems to stabilize
     * around 20% higher.
     *
     * Penetration is set to 0.05 and center of the cube is placed at (0,0.25-0.25,0.0)
     *
     * Force is along Y axis.
     * 
     * Convergence test were performed on similar case with sphere
     * https://github.com/simbody/simbody/blob/master/Simbody/tests/TestElasticFoundationForce.cpp
     */
    void testEffBrickOnPlaneNewFormulation()
    {
        // Global stiffness of the contact: each material will have
        // twice this stiffness to obtain this global stiffness in the contact
        // 1/kG = 1/k1 + 1/k2
        const Real stiffness = 100.0;
        const Real dissipation = 0.0, us = 0.0, ud = 0.0, uv = 0.0;
        const Real vt = 1.0;
        const int maxLevel = 30;
        const Real penetration = 0.05;
        for (int i=1;i<maxLevel;++i)
        {
            // Creation of the classical problem
            MultibodySystem system;
            SimbodyMatterSubsystem matter(system);
            ContactTrackerSubsystem tracker(system);
            CompliantContactSubsystem contactForces(system, tracker);
            contactForces.setTransitionVelocity(vt);
            matter.Ground().updBody().addContactSurface(
                Transform(Rotation(-0.5*Pi, ZAxis), Vec3(0.0,0.0,0.0)),
                ContactSurface(ContactGeometry::HalfSpace(),
                               ContactMaterial(2.0*stiffness, dissipation, us, ud, uv),
                               1.0));
            Body::Rigid body(MassProperties(1.0, Vec3(0), Inertia(1)));
            body.addContactSurface(Transform(),
                ContactSurface(ContactGeometry::TriangleMesh(PolygonalMesh::createBrickMesh(Vec3(0.25,0.25,0.25), i)),
                               ContactMaterial(2.0*stiffness, dissipation, us, ud, uv),
                               1.0));
            MobilizedBody::Free mesh(matter.updGround(), Transform(), body, Transform());
            State state = system.realizeTopology();
            mesh.setQToFitTransform(state, Transform(Vec3(0.0,0.25-penetration,0.0)));
            system.realize(state, Stage::Dynamics);
            ASSERT(contactForces.getNumContactForces(state)==1);
            const ContactForce& force = contactForces.getContactForce(state,0);
            const Vec3& frc = force.getForceOnSurface2()[1];
            // cout << force.getForceOnSurface2() << endl;
            const Real numericalResult = frc[1];
            const Real intersectionVolume = penetration * 0.5 *0.5;
            const Real theoreticalResult = stiffness * intersectionVolume;
            cout << "Resolution " << setw(3) << i << " Error = " << 100 * abs((numericalResult/theoreticalResult)-1.0) << "%" << endl;
        }
    }

    int main() {
        try {
            testEffBrickOnPlaneNewFormulation();
        }
        catch(const std::exception& e) {
            cout << "exception: " << e.what() << endl;
            return 1;
        }
        cout << "Done" << endl;
        return 0;
    }

The program produces:

    Resolution   1 Error = 2.22045e-014%
    Resolution   2 Error = 3.33067e-014%
    Resolution   3 Error = 8.33333%
    Resolution   4 Error = 13.3333%
    Resolution   5 Error = 14.8148%
    Resolution   6 Error = 16.3265%
    Resolution   7 Error = 18.75%
    Resolution   8 Error = 19.7531%
    Resolution   9 Error = 20%
    Resolution  10 Error = 19.8347%
    Resolution  11 Error = 19.4444%
    Resolution  12 Error = 18.9349%
    Resolution  13 Error = 19.0476%
    Resolution  14 Error = 19.2593%
    Resolution  15 Error = 19.2708%
    Resolution  16 Error = 19.3772%
    Resolution  17 Error = 19.7531%
    Resolution  18 Error = 19.9446%
    Resolution  19 Error = 20%
    Resolution  20 Error = 19.9546%
    Resolution  21 Error = 19.8347%
    Resolution  22 Error = 19.6597%
    Resolution  23 Error = 19.6759%
    Resolution  24 Error = 19.7333%
    Resolution  25 Error = 19.7239%
    Resolution  26 Error = 19.7531%
    Resolution  27 Error = 19.898%
    Resolution  28 Error = 19.9762%
    Resolution  29 Error = 20%

sherm1, 2018-06-18:
Great study, Guillaume - thanks! That result is a surprise to me and definitely needs more investigation. The convergence to 20% off is odd -- clearly the algorithm thinks that's the answer. But I don't know why. Any thoughts @chrisdembia, @tkuchida, or @aseth1? Taking a close look at the generated mesh might be helpful.

Gjacquenot, 2018-06-18:
Thanks Michael !
I discover this behaviour with my own refined meshes where i subdivided
each triangle into four new triangles. So first thing tout do is to check
meshes but soon we will need other ideas!

Gjacquenot, 2018-06-19:
I have created a test program to better expose the conditions of divergence.
It sweeps different penetrations and resolution, and exports the error as a matrix.

I suspect that errors start to appear when there are some vertical triangles of the cube that are fully under the plane.

    #include "SimTKsimbody.h"
    #include <vector>

    using namespace SimTK;
    using namespace std;

    const Real TOL = 1e-5;

    #define ASSERT(cond) {SimTK_ASSERT_ALWAYS(cond, "Assertion failed");}

    template <class T>
    void assertEqual(T val1, T val2) {
        ASSERT(abs(val1-val2) < TOL || abs(val1-val2)/max(abs(val1), abs(val2)) < TOL);
    }

    template <int N>
    void assertEqual(Vec<N> val1, Vec<N> val2) {
        for (int i = 0; i < N; ++i)
            assertEqual(val1[i], val2[i]);
    }

    /**
     * Program to test convergence of EFF algorithm from collision of a box on a
     * plane. Geometries involved are a plane `ContactGeometry::HalfSpace()` and
     * a cube `PolygonalMesh::createBrickMesh`
     *
     * Analytical solution is stiffness * penetration * area_of_contact
     *
     * Numerical solution can thus be compared to the analytical one.
     *
     * A cube of length 0.5 is created with `createBrickMesh(Vec3(0.25,0.25,0.25), i)`
     * where is the resolution.
     *
     * Resolution is progressively increased from 1 to 60. Result are exact for
     * resolution=1 and 2, but increased with resolution and seems to stabilize
     * around 20% higher.
     *
     * Penetration is set to 0.05 and center of the cube is placed at (0,0.25-0.05,0.0)
     *
     * Force is along Y axis.
     *
     * Convergence test were performed on similar case with sphere
     * https://github.com/simbody/simbody/blob/master/Simbody/tests/TestElasticFoundationForce.cpp
     */
    std::vector<Real> testEffBrickOnPlaneNewFormulation(Real penetration, int maxLevel = 15);
    std::vector<Real> testEffBrickOnPlaneNewFormulation(Real penetration, int maxLevel)
    {
        // Global stiffness of the contact: each material will have
        // twice this stiffness to obtain this global stiffness in the contact
        // 1/kG = 1/k1 + 1/k2
        const Real stiffness = 100.0;
        const Real dissipation = 0.0, us = 0.0, ud = 0.0, uv = 0.0;
        const Real vt = 1.0;
        std::vector<Real> errors;
        for (int i=1;i<maxLevel;++i)
        {
            MultibodySystem system;
            SimbodyMatterSubsystem matter(system);
            ContactTrackerSubsystem tracker(system);
            CompliantContactSubsystem contactForces(system, tracker);
            contactForces.setTransitionVelocity(vt);
            matter.Ground().updBody().addContactSurface(
                Transform(Rotation(-0.5*Pi, ZAxis), Vec3(0.0,0.0,0.0)),
                ContactSurface(ContactGeometry::HalfSpace(),
                            ContactMaterial(2.0*stiffness, dissipation, us, ud, uv),
[... 36 lines cut ...]
                for(size_t j=0;j<errors[i].size()-1;++j)
                    cout << errors[i][j] << ",";
                cout << errors[i][errors[i].size()-1];
                cout << endl;
            }
        }
        else
        {
            double penetration = atof(argv[1]);
            testEffBrickOnPlaneNewFormulation((Real)penetration);
        }

        return 0;
    }

It returns the following array

    Z = np.array([[8.88178e-14,1.11022e-13,8.88178e-14,8.88178e-14,2.22045e-14,6.66134e-14,1.66533e-13,2.66454e-13,8.88178e-14,3.10862e-13,1.88738e-13,2.22045e-13,6.88338e-13,2.33147e-13],
    [8.88178e-14,1.11022e-13,8.88178e-14,8.88178e-14,2.22045e-14,6.66134e-14,1.88738e-13,1.9984e-13,4.44089e-14,2.22045e-13,2.9976e-13,2.66454e-13,7.32747e-13,5.66214e-13],
    [8.88178e-14,6.66134e-14,2.22045e-14,8.88178e-14,2.44249e-13,1.11022e-13,2.22045e-14,3.10862e-13,4.44089e-14,2.33147e-13,2.33147e-13,4.21885e-13,5.44009e-13,5.10703e-13],
    [8.88178e-14,6.66134e-14,2.22045e-14,8.88178e-14,2.44249e-13,1.11022e-13,0,3.10862e-13,6.66134e-14,2.22045e-13,2.22045e-13,4.44089e-13,5.32907e-13,5.32907e-13],
    [2.22045e-14,1.11022e-14,1.11022e-14,2.22045e-14,1.77636e-13,4.44089e-14,2.22045e-14,1.64609,3.33333,4.40771,5.09259,5.52268,5.78231,5.92593],
    [2.22045e-14,3.33067e-14,8.33333,13.3333,14.8148,16.3265,18.75,19.7531,20,19.8347,19.4444,18.9349,19.0476,19.2593],
    [16.6667,29.6296,37.5,40,38.8889,38.0952,38.5417,39.5062,40,39.6694,39.3519,39.4477,39.7959,40],
    [75,77.7778,77.0833,80,78.7037,79.5918,79.6875,79.4239,80,79.6143,79.8611,79.8817,79.7619,80]])