simbody/simbody #638: Python interface: Directly working with state's Y value leads to crash

janEbert, 2018-07-04:
Using an OpenSim environment in Python, the program crashes when attempting to do anything with the state's Y value except for copying it.

Setup:
```python
from osim.env import ProstheticsEnv
env = ProstheticsEnv(visualize=False)
env.reset()
```
What works:
```python
state = env.osim_model.get_state()
print(state.getY())
```
What does not work:
```python
```

Sorry if this belongs in the [OSim-RL](https://github.com/stanfordnmbl/osim-rl) repository.

chrisdembia, 2018-07-04:
Thanks for reporting this. Do you have a backtrace?

janEbert, 2018-07-04:
There is no backtrace, the Python process crashes ("Windows is searching for an error") and the command prompt returns to normal when I close the process. Sorry for not mentioning that.

chrisdembia, 2018-07-12:
I'm a little confused: did you run the "code that does not work" after having already run "what works"? Can you share the complete code that works and the complete code that does not work?

> Sorry if this belongs in the OSim-RL repository.

The best place for this is https://github.com/opensim-org/opensim-core, because that's where the bindings for OpenSim and Simbody are created.

chrisdembia, 2018-07-12:
@janEbert, let's continue discussion at https://github.com/opensim-org/opensim-core/issues/2214

closed 2018-07-12 by chrisdembia