simbody/simbody #639: Improve clarity of `takeOneStep()` exception

chrisdembia, 2018-07-12:
It is somewhat common to get the following exception message:

```
std::exception in 'SimTK::State const & OpenSim::Manager::integrate(double)':
SimTK Exception thrown at AbstractIntegratorRep.cpp:428:
  Integrator step failed at time 0.49668703558495564 apparently because:
SimTK Exception thrown at AbstractIntegratorRep.cpp:547:
  Error detected by Simbody method AbstractIntegrator::takeOneStep(): Unable
to advance time past 0.496687.
  (Required condition 't1 > t0' was not met.)
```

I think the fix is often to use a tighter integration accuracy. Perhaps we could add that suggestion to the exception message.

This was originally reported at https://github.com/stanfordnmbl/osim-rl/issues/138#issuecomment-404577929 by @janEbert.

sherm1, 2018-07-12:
Good idea!

artin-f, 2018-11-25:
The root cause of this issue seems to be that the integration step size falls below the `Real` type's precision.  As a result, `t0 + currentStepSize` becomes indistinguishable from `t0`, and the `t1 > t0` check fails.

This can be illustrated by using very loose accuracies in the following 3 examples.    

Example | Accuracy | `t1 > t0` fails at `t =` | step size falls below precision at `t =`
-|-|-|-
ExampleGears     | 0.7 | 196.75312283552003 | 196.75312283552003
ExampleKneeJoint | 0.5 | 2.7475657411302921 | 2.7475657411302903
Double pendulum (from the User Guide) | 0.6 | 22.839836829814722 | 22.839836829814306

The solution could then be to:
1. Replace the `t1 > t0` check with a step size check, and
2. Throw a corresponding exception that suggests using a tighter accuracy (as @chrisdembia suggested).  

There are more steps necessary than the two above, but I'll leave that for the PR discussions.  For now, I wanted to ask if you agree with the root cause.