simbody/simbody #640: investigation about contact dynamics computation

jacknlliu, 2018-08-03:
Does simbody have efficient contact dynamics implement like something provided by [SimBenchmark](https://leggedrobotics.github.io/SimBenchmark/)?

sherm1, 2018-08-03:
Simbody supports compliant contact but not rigid so may not be directly comparable to those engines. Performance depends a lot on stiffnesses and choice of numerical integration method, of which there are several. There are some contact examples in the Simbody repo, including legged locomotion (a Theo Jansen Strandbeest). This [paper](https://simtk.org/docman/view.php/47/1589/Sherman-2011-SethDelp-Simbody-ProcediaIUTAM-v2-p241.pdf) describes the Simbody contact models in section 5.

jacknlliu, 2018-08-04:
@sherm1 thanks very much for your reply. I have read the paper, and it's very useful for understanding simbody. And did simbody support rigid contact model now?

sherm1, 2018-08-04:
No. There is some preliminary support for unilateral constraints but mostly in prototype form. Simbody is intended for scientific computing so we have been concerned about accuracy, something that is hard to obtain with rigid models. Work in that direction led to [this paper](http://rspa.royalsocietypublishing.org/content/471/2177/20140859) on impacts.

jacknlliu, 2018-08-05:
@sherm1 Thanks very much for your reply! It's very valuable.

closed 2020-02-20 by jacknlliu