simbody/simbody #641: Deleting a CPodesIntegrator leads to a crash

carmichaelong, 2018-08-23:
The following code crashes:

```    
CPodesIntegrator* integPtr = new CPodesIntegrator(sys);
delete integPtr;
```

This does not occur with any other integrator.

sherm1, 2018-08-23:
Thanks, Carmichael! That definitely needs investigating. Any volunteers?

carmichaelong, 2018-08-27:
I dug around a little more. Looks like the memory is allocated by [this method](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/sundials/src/cpodes/cpodes.c#L340) which is called by [`CPodes::init()`](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/SimTKcpodes.cpp#L292). So, the crash occurs since initialization hasn't occurred. Further testing seemed to confirm this as the following code did not cause a crash:
```
CPodesIntegrator* integPtr = new CPodesIntegrator(sys);
TimeStepper ts(sys);
ts.setIntegrator(*integPtr);
ts.initialize(sys.getDefaultState());
delete integPtr;
```

chrisdembia, 2018-08-28:
@carmichaelong I'm happy to look at this with you briefly tomorrow.

carmichaelong, 2018-08-28:
@chrisdembia I'll be gone till Tuesday but happy to work on it then. If you wanted to look at it before then, here are some notes:

- There are related "Alloc" and "Free" functions that control memory management. You can see their prototypes [here](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/sundials/src/cpodes/cpodes.c#L92).
- The problem is with the call to `cpFreeVectors()` [here](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/sundials/src/cpodes/cpodes.c#L1618).
- The problem goes away if it's initialized because `cpAllocVectors()` is called within `CPodeInit()` [here](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/sundials/src/cpodes/cpodes.c#L417).

Two (rough) ideas about how to go about dealing with this:
1. Fix the underlying problem, which is there doesn't seem to be a flag specifically for when the Vectors have been allocated memory. In other cases, there are corresponding flags for whether or not memory has been allocated (see [`CPodeFree()`](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/sundials/src/cpodes/cpodes.c#L1610) and [`CPodeQuadFree()`](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodes/sundials/src/cpodes/cpodes.c#L1644)).
2. Without touching the underlying CPODES code, make changes to the SimTK interface. Specifically, make sure to call `CPodeInit()` on construction rather than in [initialization here](https://github.com/simbody/simbody/blob/master/SimTKmath/Integrators/src/CPodesIntegrator.cpp#L184). This specific call, then, would use `reInit()` instead.