simbody/simbody #644: 3.6.1 failed test TestCustomConstraints

apalazzi, 2018-10-10:
Hi,

test TestCustomConstraints fails with 3.6.1 on ubuntu 18.04/gcc 7.3.0:

```
Starting test TestCustomConstraints ...
  testCoordinateCoupler1 ...
  testCoordinateCoupler1 done. real/CPU ms: 25.3 / 18.7
  testCoordinateCoupler2 ...
  testCoordinateCoupler2 done. real/CPU ms: 304.8 / 169.9
  testCoordinateCoupler3 ...
  testCoordinateCoupler3 done. real/CPU ms: 177.4 / 89.8
  testSpeedCoupler1 ...
  testSpeedCoupler1 done. real/CPU ms: 7.1 / 1.3
  testSpeedCoupler2 ...
  testSpeedCoupler2 done. real/CPU ms: 2118.3 / 1058.4
Test failed due to exception: SimTK Exception thrown at TestCustomConstraints.cpp:451:
  Internal bug detected: Test values should have been numerically equivalent at tolerance=1e-06.
  (Assertion 'SimTK::Test::numericallyEqual((0.0),(power),1,(10*integ.getConstraintToleranceInUse()))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

TestCustomConstraints done. real/CPU ms: 2633.9 / 1339.1
```

chrisdembia, 2018-10-10:
What platform, compiler, and LAPACK/BLAS are you using?

apalazzi, 2018-10-10:
```
andrea@navicella:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic

andrea@navicella:~$ gcc --version
gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

andrea@navicella:~$ apt show libblas-dev|grep Version:
Version: 3.7.1-4ubuntu1
```

moorepants, 2018-10-10:
Note that this is commonly reported: https://github.com/search?q=TestCustomConstraints&type=Issues

traversaro, 2018-11-03:
I have the same issue with the current master branch on a stock Ubuntu 18.04 . 
I added a small print at line 450:
~~~cpp
        std::cerr << "power is " << power << " " << " tolerance is " << 10*integ.getConstraintToleranceInUse() << std::endl;
~~~
and the relevant output is attached to the issue. 
[simbody-TestCustomConstraint-371b820d3df7f6f38017cd55536ab5d08beb5919.txt](https://github.com/simbody/simbody/files/2544892/simbody-TestCustomConstraint-371b820d3df7f6f38017cd55536ab5d08beb5919.txt)

traversaro, 2018-11-03:
Just to make a test, I tried to increase the tolerance and it seems that energy goes above `1e-06` just for a few timesteps, and then it decreases for the rest of the test: 
[simbody-TestCustomConstraint-371b820d3df7f6f38017cd55536ab5d08beb5919-tolerance-increased.txt](https://github.com/simbody/simbody/files/2544895/simbody-TestCustomConstraint-371b820d3df7f6f38017cd55536ab5d08beb5919-tolerance-increased.txt)