simbody/simbody #646: Linker error with Simbody 3.6.1

apalazzi, 2018-10-17:
I have a linker error when I compile my application with Simbody 3.6.1:

<undefined reference to "SimTK::CacheEntryInfo::recordPrerequisiteVersions(SimTK::StateImpl const&)">

Ubuntu 18.04, gcc 7.3.0; simbody compiled with -DCMAKE_BUILD_TYPE=RelWithDebInfo 

The application was running fine with V3.5.

sherm1, 2018-10-18:
That method is only defined in Debug builds (declaration [here](https://github.com/simbody/simbody/blob/master/SimTKcommon/Simulation/include/SimTKcommon/internal/StateImpl.h#L393), definition [here](https://github.com/simbody/simbody/blob/master/SimTKcommon/Simulation/src/State.cpp#L1049)). RelWithDebInfo is a _Release_ build so doesn't define that method.

This suggests that there is a main program or calling library build with Debug (and hence depending on that method) but being linked with a library built as RelWithDebInfo. You'll need a Debug library to link with your Debug main program. (Or build everything with RelWithDebInfo.)

sherm1, 2018-10-18:
I'll close this issue but please reopen if I've misdiagnosed this.

chrisdembia, 2018-10-18:
@sherm1 since I've run into this myself a few times, would it make sense to always declare and define those functions and move the `#ifndef NDEBUG` inside the functions?

sherm1, 2018-10-18:
@chrisdembia, that would leave them with a runtime cost which would be unacceptable. They could be made empty and inline for NDEBUG so they would optimize away, but I still question whether it is ever going to work to have a Debug main program linked to a Release library. It certainly won't work on Windows, because the ABI is different (at least in the standard library). I'm not sure whether it might work on Linux, but it would depend on every data structure maintaining its size exactly between Release and Debug; we don't make any guarantees about that and I don't think the libraries do either.

If Debug and Release are actually incompatible, it may be better that we get a linker failure than a hard-to-interpret runtime core dump.

chrisdembia, 2018-10-18:
Okay that makes sense. Thanks for explaining.

apalazzi, 2018-10-18:
I'm not compiling my binaries with the -g flag:

`Building file: ../src/friction.cpp
Invoking: GCC C++ Compiler
g++ -I/home/andrea/include/simbody -I"/home/andrea/src/simbody/longi/include" -O3 -pg -Wall -c -fmessage-length=0 -Wno-reorder -no-pie -MMD -MP -MF"src/friction.d" -MT"src/friction.o" -o "src/friction.o" "../src/friction.cpp" `

Moreover, I'm linking my program against the system libboost, which I believe is not compiled with-g flag.

Also, if I can't link a debug build with RelWithDebInfo, what's the use of this build type?

apalazzi, 2018-10-18:
>           I'll close this issue but please reopen if I've misdiagnosed this.

How do I reopen the issue? Do I need to create a new one?

sherm1, 2018-10-18:
There should be a "Reopen" button at the bottom.

sherm1, 2018-10-18:
> Also, if I can't link a debug build with RelWithDebInfo, what's the use of this build type?

That is just a Release build that retains debug information in the generated binaries. That can provide more information when something goes wrong since (for example) variable names are retained. But the ABI is identical to what you get with Release (that is, you can always link Release and RelWithDebInfo objects together). Debug builds can change the ABI.

sherm1, 2018-10-18:
The build command you included above is somewhat inconclusive because it doesn't show everything. I did notice it has gprof profiling included (`-pg`) -- I'm not sure what affect that has on ABI if anything. Another (random) thought is that I'm not seeing the `-std=c++11` flag which is required for Simbody 3.6 and later (for _all_ linked-together modules).

As a diagnostic, you could check whether NDEBUG is actually defined in your main program as it should be:
```c++
#ifndef NDEBUG
#error "NDEBUG should have been defined"
#endif
```
You may also want to `#define NDEBUG` (or `-DNDEBUG`) explicitly yourself to determine if that is actually the problem.

sherm1, 2018-10-18:
By the way, if the existence of these Debug-only symbols is the _only_ thing that prevents mixed Debug/Release linking with gcc (even though it still won't work on Windows), I think it would be worthwhile to implement @chrisdembia's idea above, providing "force inline" empty implementations to prevent any runtime cost. If anyone wants to volunteer to test that to verify that the mixed-link would work correctly, please let us know (and ideally submit a PR!).

apalazzi, 2018-10-18:
Compiling with -DNDEBUG solves the problem; this however should really be pointed out in the documentation, I couldn't find it anywhere. It should be defined for every file and not just in the main file.

BTW:
```
andrea@navicella:~/src/simbody/simbody$ grep -ri "NDEBUG" .|grep -i relwith
./build/CMakeCache.txt:CMAKE_CXX_FLAGS_RELWITHDEBINFO:STRING='-DNDEBUG -O2 -g   '
./build/CMakeCache.txt:CMAKE_C_FLAGS_RELWITHDEBINFO:STRING='-DNDEBUG -O2 -g   '
```

So apparently RelWithDebInfo is also built with the -g flag.

closed 2018-10-18 by apalazzi