simbody/simbody #650: Visualizer shows up as blank/black on macOS 10.14 Mojave

chrisdembia, 2018-11-13:
On macOS 10.14, the Visualizer looks shows as plain black when running `BricardMechanism`:

[image]

I tried a bunch of other examples too; same issue.

Tested on both `master` and `Simbody-3.6.1`.

@jimmyDunne have you had any issues like this on your computer?

chrisdembia, 2018-11-13:
This might provide more information: https://stackoverflow.com/questions/52509427/mac-mojave-opengl

chrisdembia, 2018-11-13:
Ah... as suggested on that page, resizing the window causes the display to work:

[image]

jimmyDunne, 2018-11-13:
I was getting a black screen as well when I was testing the tug-of-war this morning, but I thought it was because the fwdtool was failing. I checked again and got the same behavior as what you describe.

chrisdembia, 2018-11-13:
Okay great I’m working on a workaround.

chrisdembia, 2018-11-14:
Thanks for testing! Is the mojave bottle built using the macOS 10.14 SDK (Xcode 10)?

scpeters, 2018-11-14:
It's definitely using Xcode 10, as `brew doctor` would complain otherwise, and I assume it's using the 10.14 SDK. Here's the formula:

* https://github.com/osrf/homebrew-simulation/blob/master/Formula/simbody.rb#L26-L37

scpeters, 2018-11-14:
What kind of machine are you using? Is there any chance it's a laptop with a retina display? The mac mini I tested doesn't have a retina display.

chrisdembia, 2018-11-14:
Ah yes, I've been using machines with a retina display and from what I read, that is part of the issue.

scpeters, 2018-11-14:
Ok, I was wrong before; I was actually testing on a laptop with retina display (I was connected remotely, so I forgot which machine it was). The laptop is from 2012, so it has an integrated intel gpu and a discrete nvidia gpu. I also tested with a mac mini that has an intel gpu.

The black screen doesn't occur with the discrete nvidia gpu, but it occurs when the intel gpu is in use (always for the mac mini, only if I disable dynamic switching and force the intel gpu for the laptop). Resizing fixes it in both cases.

So I actually think it's related to the intel gpu, not the retina screen.

chrisdembia, 2018-11-14:
Woah; thanks for the extensive testing! All of that is with mojave? What do you think is the best fix for now? #651 is rather unsatisfying. I guess we're supposed to move to Metal but I don't see that happening.