simbody/simbody #653: TestPlugin fails

yurivict, 2018-11-15:
```
        Start  21: TestPlugin
 21/107 Test  #21: TestPlugin .....................................***Failed    0.01 sec
```

chrisdembia, 2018-11-15:
Can you give the test's log? `ctest -R TestPlugin -V`.

yurivict, 2018-11-15:
```
$ ctest -R TestPlugin -V .
UpdateCTestConfiguration  from :/usr/ports/science/simbody/work/.build/DartConfiguration.tcl
Parse Config file:/usr/ports/science/simbody/work/.build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/usr/ports/science/simbody/work/.build/DartConfiguration.tcl
Parse Config file:/usr/ports/science/simbody/work/.build/DartConfiguration.tcl
Test project /usr/ports/science/simbody/work/.build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 21
    Start 21: TestPlugin

21: Test command: /usr/ports/science/simbody/work/.build/TestPlugin
21: Test timeout computed to be: 1500
21: Starting test TestPlugin ...
21:   testPathname ...
21: Path of this executable: ''
21: Executable directory: '/'
21: Current working directory: '/usr/ports/science/simbody/work/.build/SimTKcommon/tests/'
21: Current drive letter: ''
21: Current drive: ''
21: Path separator: '/'
21: Default install dir: '/usr/local/'
21: installDir(SimTK_INSTALL_DIR,/LunchTime/snacks): '/usr/local/LunchTime/snacks/'
21: installDir(LocalAppData,SimTK): '/usr/local/SimTK/'
21: Pathname::getFunctionLibraryDirectory() is located in: /usr/ports/science/simbody/work/.build/
21:   testPathname done. real/CPU ms: 0.6 / 0.6
21: Test failed due to exception: SimTK Exception thrown at TestPlugin.cpp:780:
21:   Internal bug detected: Test condition failed.
21:   (Assertion '(pathname == dir + "myFileName.ext")' failed).
21:   Please file an Issue at https://github.com/simbody/simbody/issues.
21:   Include the above information and anything else needed to reproduce the problem.
21: 
21: TestPlugin done. real/CPU ms: 0.7 / 0.7
1/1 Test #21: TestPlugin .......................***Failed    0.01 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.02 sec

The following tests FAILED:
	 21 - TestPlugin (Failed)
Errors while running CTest
```

yurivict, 2019-08-08:
Still fails in 3.6.1

sherm1, 2019-08-08:
This needs to be debugged on FreeBSD, which is not one of Simbody's supported platforms so someone who uses FreeBSD will have to do it (@yurivict, that may be you!).

I think it would be easy to debug. The error occurs in the very last test at line 780 in TestPlugin.cpp. The pathname produced using `Pathname::getAbsolutePathnameUsingSpecifiedWorkingDirectory()` was not what was expected, using the `@` character which is supposed to expand to the directory containing the currently-executing program. First step would be to either run the test in Debug or add some prints to see what is in `pathname` and `dir + "myFileName.ext"` that makes them different. Possibly the machine-dependent implementation of `@` needs to be specialized for FreeBSD. That implementation is the method `Pathname::getThisExecutablePath()`. Here is the code (from Pathname.cpp):
```c++
string Pathname::getThisExecutablePath() {
    char buf[2048];
    #if defined(_WIN32)
        const DWORD nBytes = GetModuleFileName((HMODULE)0, (LPTSTR)buf, sizeof(buf));
        buf[0] = (char)tolower(buf[0]); // drive name
    #elif defined(__APPLE__)
        uint32_t sz = (uint32_t)sizeof(buf);
        const int status = _NSGetExecutablePath(buf, &sz);
        SimTK_ERRCHK_ALWAYS(status==0,
                "Pathname::getThisExecutablePath()",
                "2048-byte buffer is not big enough to store executable path.");
    #else // Linux
        // This isn't automatically null terminated.
        const size_t nBytes = readlink("/proc/self/exe", buf, sizeof(buf));
        buf[nBytes] = '\0';
    #endif
    return string(buf);
}
```

Note that FreeBSD will fall through to the Linux case, which likely is not the right implementation. You should add an implementation for FreeBSD, and make sure it works. Then submit a PR with the change and we can merge it into Simbody.

yurivict, 2019-08-08:
Please note how the linux code doesn't check the error of the ```readlink``` function:
```
    #else // Linux
        // This isn't automatically null terminated.
        const size_t nBytes = readlink("/proc/self/exe", buf, sizeof(buf));
        buf[nBytes] = '\0';
    #endif
```

The manpage says:
```
RETURN VALUES
     The call returns the count of characters placed in the buffer if it
     succeeds, or a -1 if an error occurs, placing the error code in the
     global variable errno.
```
-1 would cause memory corruption.

yurivict, 2019-08-08:
This fixed the problem: https://github.com/simbody/simbody/pull/672

yurivict, 2019-08-08:
Please note how the linux code doesn't check the error of the ```readlink``` function:
```
    #else // Linux
        // This isn't automatically null terminated.
        const size_t nBytes = readlink("/proc/self/exe", buf, sizeof(buf));
        buf[nBytes] = '\0';
    #endif
```

The manpage says:
```
RETURN VALUES
     The call returns the count of characters placed in the buffer if it
     succeeds, or a -1 if an error occurs, placing the error code in the
     global variable errno.
```
-1 would cause it to crash.

Please merge the above PR and after this fix the error handling there.

sherm1, 2019-08-08:
Yikes -- definitely a bug!

sherm1, 2019-08-09:
Although #672 fixed the originally-reported problem here, I have retitled this issue for the error-checking bug and will leave it open.