simbody/simbody #659: Do not call exit from shared library

morxa, 2018-11-23:
libSimTKmath calls `exit` in several places:
```
$ git grep 'exit(' .
Optimizers/src/CMAESOptimizer.cpp:    cmaes_exit(&evo);
Optimizers/src/IpOpt/IpIpoptApplication.cpp:        exit(-1);
Optimizers/src/IpOpt/IpIpoptApplication.cpp:        exit(-1);
Optimizers/src/IpOpt/IpIpoptApplication.cpp:      exit(-1);
Optimizers/src/IpOpt/IpIpoptApplication.cpp:      exit(-1);
Optimizers/src/IpOpt/IpIpoptApplication.cpp:      exit(-1);
Optimizers/src/c-cmaes/cmaes.c:void   cmaes_random_exit(cmaes_random_t *);
Optimizers/src/c-cmaes/cmaes.c:void cmaes_readpara_exit(cmaes_readpara_t *);
Optimizers/src/c-cmaes/cmaes.c:                     "; use cmaes_exit() first",0,0);
Optimizers/src/c-cmaes/cmaes.c:cmaes_exit(cmaes_t *t)
Optimizers/src/c-cmaes/cmaes.c:} /* cmaes_exit() */
Optimizers/src/c-cmaes/cmaes.c:cmaes_random_exit(cmaes_random_t *t)
Optimizers/src/c-cmaes/cmaes.c:void cmaes_readpara_exit(cmaes_readpara_t *t)
Optimizers/src/c-cmaes/cmaes.c:  exit(1);
Optimizers/src/c-cmaes/cmaes.c:      exit(1);
Optimizers/src/c-cmaes/cmaes_interface.h:void cmaes_exit(cmaes_t *);
Optimizers/src/c-cmaes/docfunctions.txt:cmaes_exit(cmaes_t *) releases the dynamically allocated memory, 
Optimizers/src/c-cmaes/docfunctions.txt:             the execution by calling exit(1). This behavoir can 
tests/adhoc/nlpqlp.cpp:            exit(0);
tests/adhoc/nlpqlp.cpp:                exit(0);
tests/adhoc/nlpqlp.cpp:                exit(0);
tests/adhoc/nlpqlp.cpp:                    exit(0);
```

Calling `exit` in a shared libary is generally considered bad practice, as the caller has no way to catch the call, and you will be exiting the main application (the caller). It's also a blocker for getting this package into Fedora.

I would suggest to replace all `exit` calls by throwing an exception. If the exception is not caught, the behavior will be the same as it is now, but a caller has the option to catch the exception and react appropriately.

chrisdembia, 2018-11-23:
These are mostly all in dependencies that we copied into the source code.

sherm1, 2018-11-23:
We could fix them though since we do have our own copy. Anyone want to submit a PR?