simbody/simbody #665: Benchmark Testing

acxz, 2019-03-31:
There are many physics engines out there for the simulation/robotics community. As such, there is a need for benchmarking various engines for accurate and objective comparisons. Benchmarking also helps end users determine which is the appropriate engine for their task and for developers to help debug and better their own engines.

There is ongoing work for benchmarking at [SimBenchmark](https://github.com/leggedrobotics/SimBenchmark). As of now RaiSim, Bullet, ODE, MuJoCo, and DART have been benchmarked ([results](https://leggedrobotics.github.io/SimBenchmark/)), but it would be great to have more engines benchmarked.

As such, I would like to invite members of the Simbody community to help test & benchmark Simbody under the same framework. Based on my (limited) knowledge, this is the first time such benchmarking has been openly available. Utilizing this instead of performing closed benchmarking is beneficial for both the robotics community as well as the Simbody community.

sherm1, 2019-03-31:
I am always a bit suspicious of benchmarks hosted by authors of one of the packages being benchmarked -- somehow the host's package invariably comes out ahead! Simbody is open source (unlike RaiSim) and anyone is welcome to benchmark it and hopefully improve it also.

closed 2019-03-31 by sherm1

acxz, 2019-03-31:
I very much agree with that sentiment. That is why when I found out about an open-sourced benchmark tool, I was very happy. I have not seen any other packages open-source their benchmark software, which is why I believe this is a great tool to the community.