simbody/simbody #668: Mojave calculation result differs

Luli2020, 2019-05-30:
We are trying to upgrade our development system into Mojave, however we found that the calculation results differs (slightly but noticeable) from that in other platform (High Sierra or Ubuntu) so our unit test failed. Anybody has same experience or any insight about it? 
The program used MultibodySystem, TimeStepper, RungeKuttaMersonIntegrator etc., and the change happens when first TimeStepper::stepTo() function was called.

sherm1, 2019-05-31:
Can you provide more information? Which unit test(s) failed (some are touchier than others)? It would also be interesting to know whether the test(s) fail only with full optimization on, or whether it always fails with Mojave even in unoptimized builds (Debug, for example).

Luli2020, 2019-05-31:
Our system is basically a multibody vehicle with 4 wheels, and our unit test is to make sure the calculation will not change much for the expected location, velocity, and acceleration, etc after some time period. The acceleration changed more which caused the changes in velocity and location. The test failed in Majave for both cases with optimization on or off.

sherm1, 2019-05-31:
Oh, I thought you meant that Simbody's unit tests failed. If you build simbody and run all its built-in tests, are those OK?

Luli2020, 2019-05-31:
Yes, there is no issue for the Simbody unit tests.

sherm1, 2019-05-31:
A few random thoughts:
- One thing to check is whether you have any indeterminacy in your code (uninitialized variables, random numbers) that could be arbitrarily different on different machines. 
- Another possibility is that you are getting roundoff-error-sided changes in accelerations perhaps due to legitimate, but different, instruction choices made by a newer compiler on Mojave. In some very-sensitive simulations tiny changes can become magnified. You may want to look carefully at the simulation to make sure it doesn't behave chaotically during the testing.

Luli2020, 2019-05-31:
Thanks for the input, will do more investigation.