simbody/simbody #674: macOS Catalina: Visualizer is 1/4 size

chrisdembia, 2019-10-20:
On macOS Catalina, the simbody-visualizer only takes up 1/4 of the window:

[image]

This might be related: https://github.com/redeclipse/base/issues/920

Given that freeglut is deprecated on macOS, we could also consider using Metal or an abstraction layer:

- https://developer.apple.com/videos/play/wwdc2018/604/?time=430
- https://developer.apple.com/documentation/metal/migrating_opengl_code_to_metal
- https://github.com/LukasBanana/LLGL
- https://github.com/KhronosGroup/MoltenVK

chrisdembia, 2019-10-20:
@peastman do you have any suggestions?

peastman, 2019-10-20:
No idea.  When you resize the window does it remain 1/4 of the window?

scpeters, 2019-10-21:
Are you using a laptop with a retina screen?

chrisdembia, 2019-10-21:
> When you resize the window does it remain 1/4 of the window?

Yes

> Are you using a laptop with a retina screen?

Yes. This happens on both a MacBookPro and an iMac.

peastman, 2019-10-21:
Sounds like it could be a bug in resolution scaling.  On the issue that referenced this one just above, someone suggested https://github.com/redeclipse/base/commit/6dce7794981eea639d16b7f3a6809f1c0c889a46 as a possible fix.

chrisdembia, 2019-10-21:
Thank you @peastman . I'll try that when I get a chance.

chrisdembia, 2019-10-25:
- https://developer.apple.com/library/archive/documentation/GraphicsImaging/Conceptual/OpenGL-MacProgGuide/EnablingOpenGLforHighResolution/EnablingOpenGLforHighResolution.html#//apple_ref/doc/uid/TP40001987-CH1001-SW4
- https://github.com/SFML/SFML/issues/353

We can either make a .app and set NSHighResolutionCapable to false, or we can edit the OpenGL functions taking pixels to use 2x values and have a higher-res visualizer.

I think the issue introduced by Catalina is that the OS now assumes taht NSHighResolutionCapable is true.

chrisdembia, 2019-10-25:
Magnum seems like a promising library: https://github.com/mosra/magnum. It's used by DART.

chrisdembia, 2019-10-27:
Here is how SDL finds the "backing scale factor": https://github.com/spurious/SDL-mirror/blob/60cae8d8affb61ec661f1eb5d5be5fed6a3c239d/src/video/cocoa/SDL_cocoaopengl.m#L362-L383

I don't know how we'd get access to the NSView when using GLUT.

PR #676 (2019-10-28): Fix visualizer scaling on macOS Catalina
This PR fixes #674  by converting simbody-visualizer into an "app bundle". Now, the visualizer is installed as `simbody-visualizer.app/Contents/MacOS`, which allows us to set the setting `NSHighResolutionCapable` to false.

I also added a stub function to help us, in the future, to better support high-resolution displays.

I tested only on Mac. Might someone volunteer to test on Windows and Linux?

@mitkof6 , would you be willing to test on Linux?

---
This change is [[image]](https://reviewable.io/reviews/simbody/simbody/676)
closed 2019-10-28 by chrisdembia