simbody/simbody #677: CUDA compile issue with "conformingAdd" and "conformingSubtract" in "Vec.h"

msdrezavand, 2019-11-04:
When using SimBody in a .cu file being compiled with cuda 10.0 and gcc/g++ 7.4.0 the following error pops up regarding "conformingAdd" and "conformingSubtract" in "Vec.h":

simbody/include/simbody/SimTKcommon/internal/Vec.h(54): error: more than one instance of overloaded function "SimTK::Impl::conformingAdd" matches the argument list:
            function template "void SimTK::Impl::conformingAdd(const SimTK::Vec<1, E1, S1> &, const SimTK::Vec<1, E2, S2> &, SimTK::Vec<1, SimTK::CNT<E1>::Result<E2>::Add, 1> &)"
            function template "void SimTK::Impl::conformingAdd(const SimTK::Vec<N, E1, S1> &, const SimTK::Vec<N, E2, S2> &, SimTK::Vec<N, SimTK::CNT<E1>::Result<E2>::Add, 1> &)"
            argument types are: (const SimTK::Vec<1, SimTK::Real, 1>, const SimTK::Vec<1, SimTK::Real, 1>, SimTK::Vec<1, SimTK::Real, 1>)
          detected during:
            instantiation of "void SimTK::Impl::conformingAdd(const SimTK::Vec<N, E1, S1> &, const SimTK::Vec<N, E2, S2> &, SimTK::Vec<N, SimTK::CNT<E1>::Result<E2>::Add, 1> &) [with N=2, E1=SimTK::Real, S1=1, E2=SimTK::Real, S2=1]"

I tried to use different compilers but did not help.

sherm1, 2019-11-04:
I'm not sure what you mean about using Simbody from Cuda. Simbody is a C++ library that would be called from a C++ main program. Have you tried the same code just with gcc 7.4.0? If you can reproduce it in C++ we can help debug it (or fix the problem).

msdrezavand, 2019-11-04:
yes, I used it before for a C++ code with gcc 7.4.0 and C++11 feature, and everything was fine. Now, I have a .cu file instead of the same .cpp file that I was including simbody before, which i want to compile with nvcc (cuda compiler). That is actually the only difference. But is has the above error with those functions.
Do I need any special flag when including simbody in a cuda file bieng compiled with cuda?

sherm1, 2019-11-04:
I don't know -- I think that's a question for nvcc support. If you find the answer, please do post it here in case anyone else runs into the same problem.

msdrezavand, 2019-11-05:
Apparently, the recursive template implementations of elementwise add and subtract (`conformingAdd` and  `conformingSubtract`) in `Vec.h` are not supported by `nvcc`. 
I removed them and substituted with `operator+(r,l)`and `operator-(r,l)` in `ResultType.h`.
The problem is fixed now, but that would be good to test the Simbody lib for `cuda` project, as well.

sherm1, 2019-11-05:
We depend on outside contributors to the Simbody open source project for support of hardware, operating systems, or compilers that we don't use ourselves. If you have a PR in mind that would allow operation under nvcc without breaking anything else, please do submit one.

836446491, 2024-11-19:
Hello, I have encountered the same issue and would like to inquire about the detailed solution.