simbody/simbody #679: Release Simbody 3.6.2

chrisdembia, 2019-12-06:
@sherm1 can I prepare a Simbody 3.6.1 release that includes the macOS Catalina fix?

sherm1, 2019-12-06:
(Fixed comment to read 3.6.2) -- yes definitely, thanks!

chrisdembia, 2019-12-06:
Oops; thanks for the correction. Great.

chrisdembia, 2019-12-07:
Actually, @sherm1 , given the new smooth contact model, what about a 3.7 minor release instead?

sherm1, 2019-12-07:
Makes sense. Would need to update the release notes a little to advertise.

chrisdembia, 2019-12-07:
Will do.

chrisdembia, 2019-12-07:
I've started the release process, and I hope to finish it on Monday.

I'd like to restructure how we handle hosting doxygen by putting it all at `simbody.github.io`, using the new repo I just created. https://github.com/simbody/simbody.github.io. I'll update the links to doxygen that are in the user's guide, etc.

chrisdembia, 2019-12-07:
Okay, doxygen for all versions is now at https://simbody.github.io.

sherm1, 2019-12-07:
Doxygen looks good on github.io. Can you say why that's better than the old way?

chrisdembia, 2019-12-08:
With the old way, we would need to create a new repository and edit the `simbody-latest-release` repository. Now, we can edit one repository. The new URL is shorter. Lastly, the landing page allows users to select the version for which they want documentation; before, users would need to navigate to a separate URL on their own.

If you prefer the old way, I can undo the changes.

sherm1, 2019-12-08:
Nope, I was just wondering what the reason for the change was. Thanks!

chrisdembia, 2019-12-08:
@sherm1, I created a release for 3.7 (https://github.com/simbody/simbody/releases/tag/Simbody-3.7) and announced the release in the forum (https://simtk.org/plugins/phpBB/viewtopicPhpbb.php?f=47&t=11304&p=0&start=0&view=).

To finish this issue, I created a PR to update the master branch after the 3.7 release.

PR #681 (2019-12-09): Update master to 3.8.
This PR updates the master branch to version 3.8.

I also updated the CONTRIBUTING.md file (syntax error and mentioning Gil Serrancoli).

Fixes #679 .

---
This change is [[image]](https://reviewable.io/reviews/simbody/simbody/681)
closed 2019-12-09 by sherm1