simbody/simbody #682: Simbody 3.7: TestCustomConstraints fails

apalazzi, 2020-01-18:
Hi,
The test TestCustomConstraints fot Simbody 3.7 fails on my system (Debian testing "bullseye" with some elements from unstable) with the following message:

```
Starting test TestCustomConstraints ...
  testCoordinateCoupler1 ...
  testCoordinateCoupler1 done. real/CPU ms: 2.2 / 24.1
  testCoordinateCoupler2 ...
  testCoordinateCoupler2 done. real/CPU ms: 38.7 / 452.2
  testCoordinateCoupler3 ...
  testCoordinateCoupler3 done. real/CPU ms: 43.8 / 506.9
  testSpeedCoupler1 ...
  testSpeedCoupler1 done. real/CPU ms: 0.9 / 11.3
  testSpeedCoupler2 ...
  testSpeedCoupler2 done. real/CPU ms: 205.9 / 364.5
Test failed due to exception: SimTK Exception thrown at TestCustomConstraints.cpp:451:
  Internal bug detected: Test values should have been numerically equivalent at tolerance=1e-06.
  (Assertion 'SimTK::Test::numericallyEqual((0.0),(power),1,(10*integ.getConstraintToleranceInUse()))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

TestCustomConstraints done. real/CPU ms: 293.9 / 1384.9
```

chrisdembia, 2020-01-18:
I think this is the same issue as #644, which you opened previously. Closing for now.

closed 2020-01-18 by chrisdembia

apalazzi, 2020-01-19:
@chrisdembia so, should I trust Simbody or not when using custom constraints? If the tolerance in the test is too strict, a proper fix should be to modify the tolerance in the test, no? Closing the bug and leaving the issue standing doesn't seems correct to me.

sherm1, 2020-06-14:
BTW the but is not closed -- this was a duplicate of #644 which is still open and the problem still exists. I _think_ this is a problem with the test case rather than custom constraints, but that has yet to be proven.