simbody/simbody #683: Different behaviour when using simbody RelWithDebInfo vs Release 3.7

apalazzi, 2020-01-18:
Hi,

I'm refactoring my program, and I have a strange behavior: if I use the (dynamic) libraries created with RelWithDebInfo it ends with an error "free(): invalid pointer", but if I run the debug version the error is different:
```
/home/andrea/simbody/simbody/SimTKmath/Geometry/include/simmath/internal/GCVSPLUtil.h:95: static SimTK::Vec<MM, double, 1> SimTK::GCVSPLUtil::splder(int, int, SimTK::Real, const Vector&, const SimTK::Vector_<SimTK::Vec<MM, double, 1> >&) [with int K = 1; SimTK::Real = double; SimTK::Vector = SimTK::Vector_<double>]: Assertion `t >= x[0] && t <= x[x.size()-1]' failed.
```
I tested different build options (-Ox, -gx, -DNDEBUG) but the result is always the same: the program using RelWithDebInfo aborts on free() while with Debug it fails the assertion.

Let me know if you need me to test something specific.

chrisdembia, 2020-01-18:
Are you building your client project with the same flags as those with which you compiled Simbody?

apalazzi, 2020-01-19:
Yes, same flags. I've also used the _d libraries on the release build and the release libraries with the debug build with the same result: assert with the debug libraries and invalid pointer(*) with the release libraries.

apalazzi, 2020-01-19:
I didn't specify any flags while compiling Simbody, the build system did all the job; moreover neither the readme nor the document linked at "you want to... use simbody in your own program" - aka SimbodyAndMolmodelUserGuide.pdf - says anything about flags or defines to be used during compilation of my program.