simbody/simbody #685: Support variable mass properties

chrisdembia, 2020-02-27:
@nbianco and I are hoping to support optimizing a body's mass in Moco. Right now, changing a body's mass require's invoking `OpenSim::Model::initSystem()` which is too expensive to do during direct collocation.

I had hoped that Simbody would allow editing the mass properties in the state (at stage Instance). I found the following in the source code:

https://github.com/simbody/simbody/blob/c6f465bc7de0ca2adaecdbffac6076a173f5a347/Simbody/src/MobilizedBodyImpl.h#L257-L261

This indicates that Simbody does not yet support updating mass properties after `realizeTopology()`.

@sherm1, do you have a sense of how difficult it would be implement a `MobilizedBody::setMassProperties(SimTK::State&)`?

chrisdembia, 2020-02-27:
@nickbianco

sherm1, 2020-02-27:
Not difficult, I think -- mostly bookkeeping to take care of. We have examples of this already. The first one that came to my mind is [Thermostat](https://simbody.github.io/3.7.0/classSimTK_1_1Force_1_1Thermostat.html) which has the standard patttern: for example, `setDefaultBathTemperature(temp)` and `setBathTemperature(State&, temp)`.

Then the code in its [realizeTopology()](https://github.com/simbody/simbody/blob/master/Simbody/src/Force_Thermostat.cpp#L438) method allocates Model and Instance stage variables that hold the runtime values for those parameters. And those get initialized from the default values.

For mass properties of MobilizedBodies I _think_ I already assumed we would want them to vary so there may be a generic instance stage variable belonging to SimbodyMatterSubsystem that holds them all and could be used to implement the variable ones (been a while since I looked at that).

sherm1, 2020-02-27:
Yeah, [this state variable](https://github.com/simbody/simbody/blob/master/Simbody/src/SimbodyTreeState.h#L1291) is what I was thinking of. That already holds runtime values for all the mass properties and other parameters. Those get initialized to the default values -- I think all you would need to do is add APIs for modifying these (and make sure they don't accidentally get overridden afterwards from the defaults).

chrisdembia, 2020-02-27:
Thanks @sherm1 that's very helpful. Maybe we'll give it a shot.