simbody/simbody #687: Use CMake's ${CMAKE_CURRENT_SOURCE_DIR} instead

HiImJulien, 2020-04-24:
Hey there!
(TL;DR: What needs to be done to fix this is on the bottom.)

CMake offers two mechanisms to fetch dependencies on the fly. The older `ExternalProject` and the newer `FetchContent`.

The way latter works is really simple:
 1. Declare dependency
 2. Check if satisfied, if not, download it
 3. Add subdirectory

The only catch comes with how a project specifies it's file paths.
In my newer projects I use `${CMAKE_CURRENT_SOURCE_DIR}/source.cc`, which causes CMake to search for the file relative to the current CMakeLists that declares the file. 

SimBody, however, uses CMAKE_SOURCE_DIR instead, which relates the paths to the top-level CMake file causing FetchContent Setups to subsequently fail on configure. 

**Solution:** Replace every path to use `${CMAKE_CURRENT_SOURCE_DIR}`, so that users can easily pull in this project using the `FetchContent` API. 

This is especially neat for users that don't have admin priviliges to install/run conda. Additionally I might have postgraduates checking out my repository, who I don't want to bother with getting a project's dependency straight.

Below is a sample of how one would use the `FetchContent` API.

```
cmake_minimum_required(VERSION 3.11)
project(sample)

FetchContent_Declare(
    simbody
    GIT_REPOSITORY https://github.com/simbody/simbody.git
    GIT_TAG "Simbody-3.7"
)

FetchContent_GetProperties(simbody)
if(NOT simbody_POPULATED)
    FetchContent_Populate(simbody)
    add_subdirectory(${simbody_SOURCE_DIR})
endif()

add_executable(
    sample_app
    "${CMAKE_CURRENT_SOURCE_DIR}/src/main.cpp"
)

target_link_libraries(
    sample_app
    PRIVATE
    SimTK
)
```

moorepants, 2020-04-24:
> This is especially neat for users that don't have admin priviliges to install/run conda.

If this is an issue, note that admin priviledges are not needed to install or run conda. By default it installs into non-admin locations.

sherm1, 2020-04-24:
I'm unclear about what is being proposed here. Is there a change needed to Simbody or is this for people who want to use Simbody? Should there be a documentation change somewhere?

chrisdembia, 2020-04-25:
@HiImJulien I am okay with converting usages of `${CMAKE_SOURCE_DIR}` to `${CMAKE_CURRENT_SOURCE_DIR}` in the root `CMakeLists.txt` file in this repository.

Making this change in non-root `CMakeLists.txt` files would change behavior, correct? Outside of the root CMakeLists, you could maybe replace `${CMAKE_SOURCE_DIR}` with `${Simbody_SOURCE_DIR}`.

HiImJulien, 2020-04-26:
@moorepants I've never used conda before and just assumed that this is an requirement. Thanks for correction me on this! I'd prefer to rather not rely on external tools, when I already have a working compiler, git, cmake and VIM environment.

@chrisdembia I would change it in the root as well, which would eliminate the need for a `${Simbody_SOURCE_DIR}` completely, I suppose.

@sherm1 This request is more targeted toward the usage of CMake. Let's consider the simple 'user story' in the following:

Developer A wants to develop an application using SimBody. Thus he creates a new directory with perhaps following structure:

```
my_awesome_project
├── CMakeLists.txt
└── src
```

Now with this setup he has several choices on how to pull SimBody into the project:
- Install SimBody on his system, document its usage and version number, so that the correct version will be installed by the next developer/user; -> prone to errors
- Checkout this repository as a submodule/subtree; -> should be [avoided](https://codingkilledthecat.wordpress.com/2012/04/28/why-your-company-shouldnt-use-git-submodules/)
- Let CMake download SimBody and build the project for you; only requires a working network connetion on the initial build

Since the last approach seems the safest, he decides to go for it and thus writes following into his CMake file:

```
cmake_minimum_required(VERSION 3.11)
project(my_awesome_project)

FetchContent_Declare(
    simbody
    GIT_REPOSITORY https://github.com/simbody/simbody.git
    GIT_TAG "Simbody-3.7"
)

FetchContent_GetProperties(simbody)
if(NOT simbody_POPULATED)
    FetchContent_Populate(simbody)
    add_subdirectory(${simbody_SOURCE_DIR})
endif()

add_executable(
    my_awesome_project
    "${CMAKE_CURRENT_SOURCE_DIR}/src/main.cpp"
)

target_link_libraries(
    sample_app
    PRIVATE
    SimTK
)
```

He runs `> cmake . -Bbuild`, telling cmake to configure his project. Knowing this setup should solve all his dependency management troubles he's satisfied. 
Quickly the developer realizes, that this approach, which works for other projects, doesn't work here. Alas, he's very disapointed and decides to use a (perhaps) less suited library, which, at least, allows him to utilize it more easliy.

---
So, what's the problem that makes the developer sad? Let's take a look at the directory structure.
If we run `cmake . -Bbuild` in our directory, CMake creates a directory `CMakeFiles` with another directory in it, called `_deps`, which furthermore contains simbody; i.e. like this:
```
my_awesome_project
├── CMakeFiles
│   └── _deps
│       └── simbody_src
├── CMakeLists.txt
└── src
```
With the command `add_subdirectory(${simbody_SOURCE_DIR})` we tell CMake to include SimBody's CMakeFiles into our project to be consumed. This is were the fun begins.

If you use `${CMAKE_SOURCE_DIR}`, we tell CMake to search from the top-most CMakeLists.txt. 
In the context above, a path like `${CMAKE_SOURCE_DIR}/cmake/pkgconfig/simbody.pc.in` (found in your CMakeLists), would be evaluated to `my_awesome_project/cmake/pkgconfig/simbody.pc.in`, which is obviously wrong. What we want instead is it to be evaluated to `my_awesome_project/CmakeFiles/_deps/simbody_src/cmake/pkgconfig/simbody.pc.in`.

How can we achieve this? We use `${CMAKE_CURRENT_SOURCE_DIR}`, making the path relative to the currently evaluated CMakeLists (aka the path of your CMakeLists).

---

I hope this clarifies the issue and why this, yet so minor change, is required.

P.S.: English is not my first language and I am trying to improve my explanations. If there is anything unclear, please let me know.

sherm1, 2020-04-27:
Thanks -- very helpful! And your English seems perfect.

One minor clarification: it's "Simbody" without the capital "B".

HiImJulien, 2020-04-27:
@sherm1 Thanks, I'll keep that on my mind! 

Lately an idea snuck up in my head: Why not create a fork, fix this issue and update all the build scripts to harness the power of modern CMake?

sherm1, 2020-04-27:
And then submit a PR back to the master branch? Great idea!

One difficulty with major changes to the build scripts is that they must be retested carefully on all platforms. We support Windows, Mac, and Ubuntu and others have modified the build scripts to support a few other variants (FreeBSD and MingW I think?). And of course all of these systems come in lots of different versions and administrative styles. The current scripts have the advantage of having been beaten on a long time so they are at least functional, if ugly. We'd like to keep the "functional" part but drop the "ugly"!

Pratool, 2021-01-18:
@HiImJulien Have you made progress? I would like to use Simbody CMake exported targets and am willing to make changes necessary in Simbody's build system to enable this. However, I do not want to duplicate effort if you are updating all the build scripts to leverage modern CMake.

HiImJulien, 2021-04-18:
@Pratool Excuse the late answer; I am hospitalized for a while now. 

However, as soon as I'll get dismissed, I'll try to dig up my previous work. Which might prove helpful.
Also, feel free to add me as a reviewer ;)