simbody/simbody #688: Ubuntu 16.04 simbody-visualizer ExamplePendulum error

freeyyc, 2020-04-26:
Ubuntu 16.04
builded successful

when run ./ExamplePendulum. get a Error:
yyc@yyc-HP-Z4-G4-Workstation:~/simbody-source/build$ ./ExamplePendulum
Reactions @M: ~[~[~[24.7279,30.7279,-0.2],~[-33.76,33.56,-12]] ~[~[6.36396,9.36396,0],~[-16.88,16.78,-6]] ~[~[6.66134e-16,3,0],~[-7.92,5.94,-3]] ~[~[-1.18329e-30,-7.88861e-31,-4.44089e-16],~[0,0,0]] ~[~[-1.32625e-30,1.31634e-30,1.23252e-15],~[1.77636e-17,1.23252e-15,-1.03042e-30]]]
Reactions @F: ~[~[~[-24.7279,-30.7279,0.2],~[33.76,-33.56,12]] ~[~[-6.36396,-9.36396,-0],~[16.88,-16.78,6]] ~[~[-6.66134e-16,-3,-0],~[7.92,-5.94,3]] ~[~[1.18329e-30,7.88861e-31,4.44089e-16],~[-0,-0,-0]] ~[~[1.32625e-30,-1.31634e-30,-1.23252e-15],~[-1.77636e-17,-1.23252e-15,1.03042e-30]]]
norm of difference: 0
Pin mobilizer reaction forces:
FB_G=~[~[2.12132,5.12132,0.0707107],~[-16.88,16.78,-6]] ~[~[0,0,-5.94],~[-7.92,5.94,-3]]
Constraint reaction forces (should be the same):
FC_G=~[~[2.12132,5.12132,0.0707107],~[-16.88,16.78,-6]] ~[~[-2.66454e-15,3.10862e-15,-5.94],~[-7.92,5.94,-3]]
Hit ENTER to run a short simulation ...simbody-visualizer listenerThread: unrecoverable error:
SimTK Exception thrown at simbody-visualizer.cpp:2048:
  Error detected by Simbody method simbody-visualizer: An attempt to read() 1 bytes from pipe 3 failed with errno=4 (Interrupted system call).
  (Required condition 'retval!=-1' was not met.)

FM_G=~[~[~[29.3755,27.4883,-4.92352],~[-40.9974,36.0737,-12]] ~[~[8.68779,7.74405,2.66454e-15],~[-20.4986,18.0368,-6]] ~[~[2.86849,0.878508,0],~[-8.58479,9.29639,-3]] ~[~[-1.69336e-30,-2.81945e-30,-2.87144e-16],~[-4.44089e-16,0,7.88861e-31]] ~[~[-6.28915e-30,-1.91384e-30,2.69026e-16],~[1.38336e-15,9.95069e-16,5.31156e-30]]]
yyc@yyc-HP-Z4-G4-Workstation:~/simbody-source/build$

sherm1, 2020-04-30:
Thank you for reporting this. Currently I only have access to Ubuntu 18.04. I built Simbody from scratch on two different Ubuntu machines (taking all the CMake defaults) but couldn't reproduce the problem. However, I do think I've seen something like that before and I wouldn't think the Ubuntu version matters. I would love to be able to track this down but need some help reproducing it. Any idea what may have been happening that caused the problem? Is it always reproducible on your machine?

jjblum, 2020-05-29:
I also use 16.04 and this exact same error occurs every single time I execute any of the following examples:
ExampleAmysIKProblem
ExampleAssemblerPlayground
ExampleCablePath
ExampleClosedTopologyMechanism
ExampleContactPlayground
ExampleCustomConstraint
ExampleIMUTracking
ExamplePendulum
ExampleScissorLift
ExampleTwoBoxCollide

The error is always
simbody-visualizer listenerThread: unrecoverable error:
SimTK Exception thrown at simbody-visualizer.cpp:2048:
  Error detected by Simbody method simbody-visualizer: An attempt to read() 1 bytes from pipe 3 failed with errno=4 (Interrupted system call).
  (Required condition 'retval!=-1' was not met.)

I built from source, from the current master. with gcc version 5.4.0

sherm1, 2020-05-29:
Thanks, @jjblum. I don't understand why the pipe read got interrupted, but maybe that shouldn't be treated as fatal. I wonder if it would fix the problem if it simply re-issued the call after getting an errno 4? I can't try that myself because I can't reproduce the problem -- any chance you could try a retry fix to see if that helps?

jjblum, 2020-05-29:
For sure - I just found, cloned, and built this in the past few hours, so you'd have to walk me through what change you'd like to implement.

jjblum, 2020-05-29:
I dropped this the cpp file from the error and it seems to have worked.

static void readDataFromPipe(int srcPipe, unsigned char* buffer, int bytes) {
    int totalRead = 0;
    while (totalRead < bytes) {
        auto retval = READ(srcPipe, buffer + totalRead, bytes - totalRead);

        int attempts = 0;
        while (retval == -1 && errno == 4 && ++attempts < 5)
        {
          retval = READ(srcPipe, buffer + totalRead, bytes - totalRead);
        }

        SimTK_ERRCHK4_ALWAYS(retval!=-1, "simbody-visualizer",
            "An attempt to read() %d bytes from pipe %d failed with errno=%d (%s).",
            bytes - totalRead, srcPipe, errno, strerror(errno));
        // The pipe was closed, perhaps because the simulator was closed.
        // Without this check, we end up in an infinite loop.
        if (retval == 0) throw ReadingInterrupted();

        totalRead += retval;
    }
}

sherm1, 2020-05-29:
Sweet! Thanks. Want to submit a PR and earn eternal fame on the contributors list?

sherm1, 2020-06-14:
Hi, Jason -- your PR #691 to fix this is ready to merge except for one request for a comment. Can you add that so we can merge it? If you want me to do it instead, please let me know.