simbody/simbody #694: IpTNLPAdapter.cpp Build Warning

CloneDeath, 2020-06-04:
When doing a clean build of the entire project, I observed the following warning:

```
[ 29%] Building CXX object SimTKmath/sharedTarget/CMakeFiles/SimTKmath.dir/__/Optimizers/src/IpOpt/IpTNLPAdapter.cpp.o
/home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpTNLPAdapter.cpp: In member function ‘virtual bool SimTKIpopt::TNLPAdapter::GetSpaces(SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::MatrixSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::MatrixSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::MatrixSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::VectorSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::MatrixSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::MatrixSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::MatrixSpace>&, SimTKIpopt::SmartPtr<const SimTKIpopt::SymMatrixSpace>&)’:
/home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpTNLPAdapter.cpp:176:8: warning: ‘__builtin___sprintf_chk’ may write a terminating nul past the end of the destination [-Wformat-overflow=]
   bool TNLPAdapter::GetSpaces(SmartPtr<const VectorSpace>& x_space,
        ^~~~~~~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from /usr/include/c++/7/cstdio:42,
                 from /usr/include/c++/7/ext/string_conversions.h:43,
                 from /usr/include/c++/7/bits/basic_string.h:6361,
                 from /usr/include/c++/7/string:52,
                 from /usr/include/c++/7/bits/locale_classes.h:40,
                 from /usr/include/c++/7/bits/ios_base.h:41,
                 from /usr/include/c++/7/ios:42,
                 from /usr/include/c++/7/istream:38,
                 from /usr/include/c++/7/sstream:38,
                 from /usr/include/c++/7/complex:45,
                 from /home/nicholas/Projects/Simulations/simbody/SimTKcommon/./include/SimTKcommon/internal/common.h:257,
                 from /home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpTypes.hpp:12,
                 from /home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpUtils.hpp:13,
                 from /home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpNLP.hpp:12,
                 from /home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpTNLPAdapter.hpp:12,
                 from /home/nicholas/Projects/Simulations/simbody/SimTKmath/Optimizers/src/IpOpt/IpTNLPAdapter.cpp:9:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:34:43: note: ‘__builtin___sprintf_chk’ output between 120 and 129 bytes into a destination of size 128
       __bos (__s), __fmt, __va_arg_pack ());
                                           ^
```

I am using `Ubuntu 18.04.4 LTS`

Issue can be reproduced by:
```
mkdir build
cd build
cmake ..
make
```

chrisdembia, 2020-06-04:
Thanks @CloneDeath. This is vendored code that we wouldn't want to diverge from the upstream Ipopt code base. It seems unlikely to me that we would want to upgrade to the latest version of Ipopt, but perhaps we could apply a patch to fix this.

Just curious...which compiler are you using? Seems like maybe GCC 7?

sherm1, 2020-06-04:
@chrisdembia since we keep a local copy of IpOpt in the Simbody repo it might be worth a small hack to get rid of the warning. Lots of people will hit that and be worried that something is wrong.

chrisdembia, 2020-06-04:
I cannot reproduce this warning using GCC 9 on macOS.

sherm1, 2020-06-12:
Fixed in #698.

closed 2020-06-12 by sherm1