simbody/simbody #696: README example contains a horizontal plane

adamkewley, 2020-06-09:
Not a fatal issue, but just something I noticed when going through the docs.

If I compile the example in the README ("Here's some code to simulate and visualize a 2-link chain") and run it, I'm getting something that looks like this:

[image]

I'm using the simbody that's built for OpenSim (tag = `Simbody-3.7`), so it's a bit older than what's in the master branch (so this might've been fixed already).

adamkewley, 2020-06-09:
I've PRed a possible fix for this:

https://github.com/simbody/simbody/pull/697

sherm1, 2020-06-12:
Fixed by #697.

closed 2020-06-12 by sherm1