simbody/simbody #699: Test template in one header

tomlogan501, 2020-07-01:
Hi,
Would it be great to regroup all test template in one header file other than declare in each test files ?

sherm1, 2020-07-01:
I'm not sure exactly what you mean by "test template" -- can you provide some specifics?

Probably a better approach would be to convert Simbody's tests to use a standard test framework like Google Test.

tomlogan501, 2020-07-02:
I talk about this type of code 
```
#define ASSERT(cond) {SimTK_ASSERT_ALWAYS(cond, "Assertion failed");}

#define ASSERT_EQUAL(val1, val2) {ASSERT(std::abs(val1-val2) < 1e-10);}
```
Google Test is nice in ROS but if you factorize this type of macro, you don't need it and the code stay independent

sherm1, 2020-07-02:
Yes, thanks -- I agree those ad-hoc macros should be centralized along with the other testing macros in [Testing.h](https://github.com/simbody/simbody/blob/master/SimTKcommon/include/SimTKcommon/Testing.h). Please feel free to submit a PR to improve that. Per the Simbody coding guidelines their names would need to change to be prefixed with `SimTK_` to avoid conflicts. They also might require some minor generalizations if they aren't the same in all the tests (do they need different tolerances?).