simbody/simbody #701: Segmentation fault in example TheoJansenStrandbeest

dpellegr, 2020-07-22:
I have successfully compiled the `TheoJansenStrandbeest.cpp` example (linux, gcc 10.1, simbody 3.7) but I get a segfault as soon as I launch it.

Here is my gdb backtrace:

```
0x0000555555563685 in SimTK::ListOfDependents::notePrerequisiteChange (this=0x5555555dafd8, stateImpl=...)
    at /usr/include/simbody/SimTKcommon/internal/StateImpl.h:2257
2257	    for (auto ckey : m_dependents) {
(gdb) backtrace 
#0  0x0000555555563685 in SimTK::ListOfDependents::notePrerequisiteChange (this=0x5555555dafd8, stateImpl=...)
    at /usr/include/simbody/SimTKcommon/internal/StateImpl.h:2257
#1  0x0000555555562b3d in SimTK::CacheEntryInfo::invalidate (this=0x5555555daf98, stateImpl=...)
    at /usr/include/simbody/SimTKcommon/internal/StateImpl.h:331
#2  0x00005555555636b6 in SimTK::ListOfDependents::notePrerequisiteChange (this=0x55555558b2a0, stateImpl=...)
    at /usr/include/simbody/SimTKcommon/internal/StateImpl.h:2260
#3  0x00007ffff7edeec5 in SimTK::StateImpl::advanceSystemToStage(SimTK::Stage) const ()
   from /usr/lib/libSimTKcommon.so.3.7
#4  0x00007ffff7eee8d7 in SimTK::System::Guts::realizeModel(SimTK::State&) const ()
   from /usr/lib/libSimTKcommon.so.3.7
#5  0x00007ffff7eefe75 in SimTK::System::Guts::realizeTopology() const () from /usr/lib/libSimTKcommon.so.3.7
#6  0x000055555555dd20 in main () at TheoJansenStrandbeest.cpp:208
```

dpellegr, 2020-07-22:
I noted that also other examples were showing the same behavior, so I dug a bit more and I found that I was missing `-DNDEBUG` on the command line.

So this produces a working executable:
```
g++ -I/usr/include/simbody/ -DNDEBUG TheoJansenStrandbeest.cpp -lSimTKcommon -lSimTKmath -lSimTKsimbody
```

sherm1, 2020-07-22:
Thanks -- that's interesting. That turns off the asserts and expensive tests that are normally performed in Debug builds. However, it is not _supposed_ to fail in Debug (although it would likely run very slowly).

BTW, as long as you are building a Release (non-Debug) build, you should optimize to get decent performance. I would suggest `g++ -O3`.

sherm1, 2020-07-22:
Simbody's CMake system knows which compiler flags should be used. If you use CMake to produce makefiles you can compile the examples using make. (All the binaries end up in the build directory that you specify to CMake.)

dpellegr, 2020-07-23:
I installed Simbody as a package, so I did not look too much into the compilation process, I just cd to the example folder and tried to compile a few manually as I would while developing my program. Some were ok, some required the flag.

And, yes, a segfault is never _supposed_ to happen :)