simbody/simbody #702: ParallelExecutor executes in multiple threads when it should not, causes access to invalid TLS

MadCatX, 2020-09-09:
Hello everyone,

I have been seeing an odd crash in some of [molmodel](https://github.com/samuelflores/molmodel) unit tests which I eventually traced back to Simbody's `ParallelExecutor` running in multithreaded mode even when it should use the singlethreaded shortcut. The following conditions have to be met for the problem to occur:

- A `Parallel2DExecutor` task is created.
- The backing `ParallelExecutor` is set to use only one thread.
- The computed problem is of such size that the last square created by `Parallel2DExecutor` is empty (= has zero size).

This causes this condition:
      if (min(times, numMaxThreads) == 1) {

to evaluate to false because `times` is 0 and `ParallelExecutor` then enters the multithreaded loop. Since the task's TLS has already been initialized to singlethreaded operation by previous iterations of the comptutation, `finish()` method of the task then tries to access the TLS of the worker thread instead of the main thread, causing a crash.

Changing the respective check to
      if (min(times, numMaxThreads) =< 1) {
      
should fix the issue.

For the record, I used [TestDuMMForces](https://github.com/samuelflores/molmodel/blob/master/tests/TestDuMMForces.cpp) from molmodel test suite to debug this.

I also wonder if there could be a "reverse" case of this problem. For instance, if the initial `TriangleTask` solution used only one thread but the subsequent `SquareTask`s were large enough for multithreaded processing. My analysis of the code suggests that this should not happen but I'm not sufficiently familiar with the code to think through all the possibilities.

sherm1, 2020-09-09:
Thanks for the great investigation, @MadCatX -- would you like to submit a PR with a fix?

MadCatX, 2020-09-09:
If you believe that the fix is sensible and cannot break something else I can re-test it on the master branch (I use stable 3.7) and submit a PR.

sherm1, 2020-09-09:
Thanks! Ideally if times==0 it should simply return rather than creating and destroying a task that does nothing. Consider making that a separate case that just returns and then leaving the other condition as is, assuming that also addresses the problem.

MadCatX, 2020-09-09:
I just realized that this fix is not sufficient. It fixes the problem if the backing `ParallelExecutor` is set to use only one thread but it creates an "inverse" case of the problem if the executor initially uses multiple threads. It seems to me that the only reliable fix would be to force the `ParallelExecutor` to use either multithreaded or singlethreaded mode throughout its entire lifetime when it is being used from `Parallel2DExecutor`.

samuelflores, 2020-09-11:
Hi Sherm!

Thanks for the help! 

These are preparatory activities for making an online nucleic acid folding service with MMB as the backend. MadCatX is not his birth name, surprisingly enough, but i guess it's not absolutely clear he wants to be outed. Anyway MadCatX is in my lab in Stockholm doing brilliant coding.

Sam

sherm1, 2020-09-11:
Hi, Sam! Nice to see MMB getting some love. Regards, Sherm

MadCatX, 2020-09-17:
Just to keep this alive, I proposed a more robust fix for the issue in this PR (https://github.com/simbody/simbody/pull/704). Ubuntu CI builds seemed to be having some issues at the time I submitted the PR.

sherm1, 2020-09-21:
Apologies for the delay -- #704 looks good with some minor review comments, PTAL.