simbody/simbody #703: Any idea why OpenSimPartyDemoCable.cpp fails to converge?

adamkewley, 2020-09-10:
Minor issue, because this code isn't part of the main build, but it would be helpful if I could understand what's going on.

I'm currently looking into optimizing OpenSim's muscle wrapping code. One suggestion is to change the underlying implementation (effectively, [this](https://github.com/opensim-org/opensim-core/tree/master/OpenSim/Simulation/Wrap) bunchgroup of code) to use SimTK's Cable implementation.

sherm1, 2020-09-11:
Hi, Adam. I'm not sure why that fails, but I did see a comment at the top that says "THIS DOESN'T WORK YET", so I guess we knew about that at some point.

I'm not sure Simbody's cable has sufficient functionality to replace OpenSim's muscle wrapping, but it might be worth investigating. A full implementation of [research by Andreas Scholz ](https://link.springer.com/article/10.1007%2Fs11044-015-9451-1)or Ian Stavness would be a great replacement though if you want something bigger to chew on.

adamkewley, 2020-09-12:
Ah, how did I not notice the comment!

Thanks for giving this a look @sherm1 , the info about alternative methods was also interesting. I'm currently just experimenting with alternative wrapping implementations that are "off the shelf", while also trying to perf tune what's already in opensim, but I'll take a look into AS's research. It might be thst i can clean room implement it, but that's fairly unlikely because my maths is fairly weak in this research area (I'm trying to study up, but it's a long slog)

Thanks again

closed 2020-09-12 by adamkewley