simbody/simbody #706: Failed build on Windows.

jasonmeverett, 2020-12-04:
**Compiling toolchain information**:
* CMake 3.19.0-rc3
* Visual Studio 16 2019
* RelWithDebInfo

**CMake Output**:
```
CMake Deprecation Warning at CMakeLists.txt:19 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.
Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Configuring done
Generating done
```

**Snippet of Visual Studio Output** (focusing on the error):
```
Rebuild started...
1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: RelWithDebInfo x64 ------
1>Checking Build System
2>------ Rebuild All started: Project: PlatformFiles_subdir, Configuration: RelWithDebInfo x64 ------
2>Copy C:\LIB\simbody\Platform\Windows\lib_x64\libblas.dll -> C:\LIB\simbody\build\RelWithDebInfo\libblas.dll
2>Copy C:\LIB\simbody\Platform\Windows\lib_x64\libgcc_s_sjlj-1.dll -> C:\LIB\simbody\build\RelWithDebInfo\libgcc_s_sjlj-1.dll
2>Copy C:\LIB\simbody\Platform\Windows\lib_x64\libgfortran-3.dll -> C:\LIB\simbody\build\RelWithDebInfo\libgfortran-3.dll
2>Copy C:\LIB\simbody\Platform\Windows\lib_x64\liblapack.dll -> C:\LIB\simbody\build\RelWithDebInfo\liblapack.dll
2>Copy C:\LIB\simbody\Platform\Windows\lib_x64\libquadmath-0.dll -> C:\LIB\simbody\build\RelWithDebInfo\libquadmath-0.dll
2>Building Custom Rule C:/LIB/simbody/Platform/CMakeLists.txt
3>------ Rebuild All started: Project: Code - Platform Files, Configuration: RelWithDebInfo x64 ------
3>Building Custom Rule C:/LIB/simbody/CMakeLists.txt
4>------ Rebuild All started: Project: Code - SimTKcommon, Configuration: RelWithDebInfo x64 ------
4>Building Custom Rule C:/LIB/simbody/SimTKcommon/sharedTarget/CMakeLists.txt
4>About.cpp
4>NiceTypeName.cpp

...
...
...

5>Geo_Point.cpp
5>Geo_Sphere.cpp
5>Geo_Triangle.cpp
5>Geodesic.cpp
5>OBBTree.cpp
5>OrientedBoundingBox.cpp
5>gcvspl.cpp
5>About.cpp
5>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\algorithm(7417,1): error C2678: binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)
5>        with
5>        [
5>            _BidIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>
5>        ] (compiling source file C:\LIB\simbody\SimTKmath\Geometry\src\ContactGeometry_TriangleMesh.cpp)

...
...
...

31>TestScalar.cpp
29>TestPolygonalMesh.cpp
35>TestXml.cpp
16>TestArray.vcxproj -> C:\LIB\simbody\build\RelWithDebInfo\TestArray.exe
26>TestParallelExecutor.vcxproj -> C:\LIB\simbody\build\RelWithDebInfo\TestParallelExecutor.exe
11>RandomTest.vcxproj -> C:\LIB\simbody\build\RelWithDebInfo\RandomTest.exe
8>MatVecTest.vcxproj -> C:\LIB\simbody\build\RelWithDebInfo\MatVecTest.exe
5>Done building project "SimTKmath.vcxproj" -- FAILED.

```

Any help would be greatly appreciated. I can provide more information if needed.

jasonmeverett, 2020-12-04:
Update:

Looks like similar error is being worked on a different project [here](https://github.com/pytorch/pytorch/issues/48517).

jasonmeverett, 2020-12-04:
Update:

I am able to get a compiled version by modifying a line in `SimTKcommon\BigMatrix\include\SimTKcommon\internal\VectorIterator.h` by changing:

```
    ELT& operator*() {
        assert (index >= 0 && index < vectorp->size());
        return (*vectorp)[(int)index];
    }
```

to:

```
    ELT& operator*() const {
        assert (index >= 0 && index < vectorp->size());
        return (*vectorp)[(int)index];
    }
```

Not quite sure why this is required but it seems like things are working now after testing a few examples.

aymanhab, 2020-12-04:
Thanks @jasonmeverett, can you open a PR with the fix so we test on all platforms? Thank you

sherm1, 2020-12-04:
That would allow mutable access to a const object -- not a good idea! I think the fix should be at the point where this is used, not here.

jasonmeverett, 2020-12-04:
@aymanhab @sherm1 Excuse my limited expertise, Aerospace by trade and software dev by tangential dependency. This is very quickly approaching an area of C++ that makes me shiver. 😄 

I can try to find where this is being called.

jasonmeverett, 2020-12-04:
If I'm reading this correctly, part of the problem appears in `ContactGeometry_TriangleMesh.cpp`:

```
5>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\include\algorithm(7417,1): error C2678: binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)
5>        with
5>        [
5>            _BidIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>
5>        ] (compiling source file C:\LIB\simbody\SimTKmath\Geometry\src\ContactGeometry_TriangleMesh.cpp)
5>C:\LIB\simbody\SimTKcommon\BigMatrix\include\SimTKcommon/internal/VectorIterator.h(76,10): message : could be 'ELT &SimTK::VectorIterator<ELT,SimTK::VectorBase<ELT>>::operator *(void)'
5>        with
5>        [
5>            ELT=SimTK::Real
5>        ] (compiling source file C:\LIB\simbody\SimTKmath\Geometry\src\ContactGeometry_TriangleMesh.cpp)
```

aymanhab, 2020-12-04:
The call sequence in question is:
ContactGeometry::TriangleMesh::Impl::splitObbAxis
median(minExtent)
median(RandomAccessIterator start, RandomAccessIterator end)
std::nth_element(start, mid, end) // implemented on msvc side, changed in the latest update ..
...

jasonmeverett, 2020-12-04:
@aymanhab I tried a few things with no success. Am I passing a `const` object when I shouldn't be, or something vice-versa?

aymanhab, 2020-12-04:
@jasonmeverett I think what we're doing on the simbody side is perfectly fine,  passing iterators on Vector down to a method that has the signature 
void nth_element( RandomIt first, RandomIt nth, RandomIt last )
I tend to think this is a change on microsoft's side to accommodate new requirements for c++20 but I haven't figured out how to switch these changes off yet to avoid making unnecessary code changes. Will keep you posted.

nickbianco, 2020-12-04:
It seems like this was a bug with Visual Studio 16.5: https://developercommunity.visualstudio.com/content/problem/953628/c-c2100-illegal-indirection-with-stdc-latest-after.html. It was fixed in 16.7, but maybe it was reintroduced somehow.

sherm1, 2020-12-05:
@jasonmeverett and @aymanhab -- my apologies. Jason's proposed fix is correct. VectorIterator is a not a const_iterator, so even when the iterator is const it can still point to a writable element. (Like a pointer declared `int* const p;`)

I have submitted PR #707 to fix this problem and also a couple of warnings I saw. Jason, can you review that for me please and ideally try it out before I merge it? (Just click the purple Reviewable button in the PR.)

PR #707 (2020-12-05): Fix compilation error and some warning in Visual Studio 16.8.2.
Visual Studio 2019 16.8.2 introduced a compilation error, fixed here. Might be a legitimate complaint.
There were also two new warnings which look spurious but are also fixed here.

Resolves #706.

cc @jasonmeverett @aymanhab

---
This change is [[image]](https://reviewable.io/reviews/simbody/simbody/707)
closed 2020-12-05 by aymanhab