simbody/simbody #709: Anaconda Windows Install - Visualizer Exception

jasonmeverett, 2020-12-17:
After installing with miniconda3 and building an example program, I get the following error message pertaining to `VisualizerProtocol`:

```
EXCEPTION OCCURED:

SimTK Exception thrown at VisualizerProtocol.cpp:165:
  Error detected by Simbody method VisualizerProtocol::ctor(): Unable to spawn executable 'simbody-visualizer' from directories:
  c:\LIB\modeller\build\Debug\
  D:/bld/simbody_1591833882663/_h_env/Library/bin/
  C:\Program Files\Simbody\bin\
  C:\Program Files\SimTK\bin\
```

Strange thing is, `simbody-visualizer` exists on the path, but it's not in any of the above directories since I am using the conda installation. It almost seems as if there's a hardcoded `C:\Program Files\Simbody\bin\` somewhere and that's where it's trying to access the visualizer.

Has anyone had success with Windows 10 (build version `20H2`), with Simbody's Anaconda packages?

I can provide more information if need be. Thanks in advance.

jasonmeverett, 2020-12-17:
To add on to this issue, it looks like the default install tests are also failing. Running PowerShell in Administrator mode

```
conda activate eng
cd C:\ProgramData\Miniconda3\envs\<my environment>\Library\examples\bin
.\SimbodyInstallTest.exe
```

Running the above for me still fails. However, running `.\SimbodyInstallTestNoViz.exe` runs completely fine and prints the appropriate output. So again, routes back to something about the visualizer install on anaconda.

moorepants, 2020-12-17:
The issue is more than likely associated with the specific build we do for conda, which is defined here:

https://github.com/conda-forge/simbody-feedstock/

Maybe we need some tests for the visualizer on windows there. I think it currently only tests if files were located in the correct place.

moorepants, 2020-12-17:
This `  D:/bld/simbody_1591833882663/_h_env/Library/bin/` should be replaced with the correct paths on your system by conda.

I think this https://github.com/conda-forge/simbody-feedstock/blob/master/recipe/meta.yaml#L18 is supposed to do that.

moorepants, 2020-12-17:
Also you say "anaconda" package, but there are no anaconda packages. You are installing from the conda forge channel, right?

jasonmeverett, 2020-12-17:
> Also you say "anaconda" package, but there are no anaconda packages. You are installing from the conda forge channel, right?

Apologies, yes, I'm referring to the conda forge channel.

jasonmeverett, 2020-12-17:
Do these paths exist anywhere in a config file on my local machine that I can "hack" update, without working through the feedstock pipeline?

moorepants, 2020-12-17:
The package is installed in in your miniconda directory. You can grep the files for that build path and change them manually.

moorepants, 2020-12-17:
This function:

https://github.com/simbody/simbody/blob/dda623d3e099b38ef9e551a6b647cb5e8dfbbb25/Simbody/Visualizer/src/VisualizerProtocol.cpp#L107

is obtaining an incorrect path. On install conda is supposed to replace all instances of `D:/bld/simbody_1591833882663/_h_env` with the correct path on the system it is installed on. We need to figure out why this isn't being replaced.

moorepants, 2020-12-17:
The path is found with this code:

https://github.com/simbody/simbody/blob/dda623d3e099b38ef9e551a6b647cb5e8dfbbb25/Simbody/Visualizer/src/VisualizerProtocol.cpp#L292

moorepants, 2020-12-17:
Do you have that path hardcoded in your cmake files?

jasonmeverett, 2020-12-17:
> Do you have that path hardcoded in your cmake files?

Nope no hard-coded path. I call CMake from PowerShell prompt after activating my environment with `conda activate`. Here's the CMake file I'm using:

```
# Generic CMakeLists.txt for making a Simbody-using executable.
# This shows how to use the provided SimbodyConfig.cmake to locate a Simbody
# installation on your machine so you can use it from your own code.
# You will most likely want to copy some of these lines into your own
# CMakeLists.txt rather than use this one verbatim.
cmake_minimum_required(VERSION 2.8)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED True)
set(CMAKE_BUILD_TYPE Release)

project(myexe)

# List your source and header files here.
set(my_source_files myexe.cpp)

# This depends on SimbodyConfig.cmake being located somewhere predictable
# on your machine. If you have installed it somewhere that CMake won't be
# able to guess, you'll need to tell find_package where to look.
find_package(Simbody REQUIRED)

# This is only necessary if Simbody was built using CMake older than 3.0.2.
include_directories(${Simbody_INCLUDE_DIR})

add_executable(myexe ${my_source_files})
target_link_libraries(myexe ${Simbody_LIBRARIES})

```

jasonmeverett, 2020-12-18:
Well, as a hacky solution here,  I ended up creating the `C:\Program Files\Simbody\bin` directory that simbody is looking for, and manually copied `simbody-visualizer.exe` from `<conda env>\Library\bin\` into this new folder. Now the visualizer opens and there's no issue. 

Considering we'd like this solution to be user-friendly I certainly don't want this to be the only solution, but I guess this works for now.

jasonmeverett, 2021-01-18:
Closing because the above solution technically works fine and is not that hard to configure. Also moved to a Docker configuration for Windows so I'm moving away from Conda install.

moorepants, 2021-01-18:
This should stay open or be transferred to the feedstock repository. It is an issue that should be fixed.

closed 2021-04-30 by jasonmeverett