simbody/simbody #711: Docker & Simbody Visualizer : GLXBadContext

jasonmeverett, 2021-01-17:
I've wrapped Simbody into a Docker image (Ubuntu 20.04), and I'm trying to also activate the visualizer. I've gotten other X11-apps to open up just fine from within the container. When I run the install test, I get the following error:

```
root@67ec1dd13b50:/simbody-source/build# ./SimbodyInstallTest
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  149 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  147
  Current serial number in output stream:  147
```

I'm running an RTX 2070 Super. Any help would be greatly appreciated. I can provide more information if needed.

Thanks!

jasonmeverett, 2021-01-17:
Update: I've tried multiple different build types (Debug, RelWithDebugInfo), and have also tried multiple different X-servers. I'm not familiar enough with the Visualizer source (or OpenGL for that matter) to understand this issue.

jasonmeverett, 2021-01-17:
Update: problem solved. First off I added a few extra libraries to my Docker image:

```
RUN apt-get install -y \
        mesa-common-dev \
        libgl1-mesa-dev \
        libglu1-mesa-dev \
        libxt-dev \
        mesa-utils 
```

I also updated my `LIBGL_ALWAYS_INDIRECT` to be `0`. Interestingly enough, on nearly every online forum I saw (except for 1), recommended procedure to fix similar-looking issues has been to issue the following:

```
export LIBGL_ALWAYS_INDIRECT=1
```

However in my case, this actually causes GL errors. I'm actually not sure if any of the above libraries are truly needed or if I just needed to update that `LIBGL` flag to `0`, but I figured I'd keep both solutions. I'll keep testing this tomorrow.

Leaving issue open for now just in case there's a different way to solve this that is more mainstream/recommended.

Anyways I now have (what seems to be) a functioning Docker image for Simbody that comes with everything that I plan on using, and the visualizer now works too! See Docker file here:

https://github.com/jasonmeverett/modeller/blob/main/Dockerfile

sherm1, 2021-01-17:
Great work -- congratulations! BTW the Dockerfile link gave me a 404 error.

jasonmeverett, 2021-01-17:
@sherm1 Oops, sorry about that, my repo was spawned as a private repo (I eventually plan on adding some work related stuff into this, but there's nothing proprietary on it yet). Here's the link to the repo:

https://github.com/jasonmeverett/modeller

I'd love your input on the Docker configuration when you get a chance.

sherm1, 2021-01-18:
Thanks Jason, now I can see it. Sadly I can't provide any useful commentary since I've never used Docker :cry:

closed 2021-01-18 by jasonmeverett