simbody/simbody #712: Including Differnet-Type Euler sequences (3-2-1, 1-3-1, etc.) into Quaternion Initialization

jasonmeverett, 2021-01-22:
We quite often default to 3-2-1 sequence for our work, which is the purpose of opening this. Eventually it would probably be nice to have some sort of default function that can handle any arbitrary sequence (like a character sequence of 'XYZ', 'ZYZ', etc, maybe?), but that's probably out of the scope of this.

jasonmeverett, 2021-01-22:
[Forked here, I'll post updates on this thread.](https://github.com/jasonmeverett/simbody)

sherm1, 2021-01-22:
Simbody does have utility methods supporting any Euler sequence -- see the various constructors and methods in the [Rotation ](https://simbody.github.io/3.7.0/classSimTK_1_1Rotation__.html) class.

3-2-1 doesn't uniquely define a sequence because it could be either "body fixed" (intrinsic) or "space fixed" (extrinsic) -- the Rotation methods take both the sequence and a specification of how it is to be interpreted.

Simbody's predefined Gimbal and Bushing mobilizers are defined to use a particular Euler sequence (1-2-3 body fixed, which is equivalent to 3-2-1 space fixed). Custom mobilizers that use any desired sequence can be defined, but you can also easily construct any desired sequence by using a sequence of pin joints with two massless bodies in between.

jasonmeverett, 2021-01-23:
Ah, apologies. I did not see this :). Thanks again for the quick response.

closed 2021-01-23 by jasonmeverett