simbody/simbody #715: ObservedPointFitter crashes attempting to get a reference to nullptr

MadCatX, 2021-03-25:
I wish I could be more descriptive but the problem seems a bit illusive to me.

I ran into this issue while fixing up [molmodel](https://github.com/samuelflores/molmodel)'s test suite. Some tests ([TestFitMagnesiums](https://github.com/samuelflores/molmodel/blob/master/tests/TestFitMagnesiums.cpp), [TestLoadPdb](https://github.com/samuelflores/molmodel/blob/master/tests/TestLoadPdb.cpp), [TestProteinDihedrals](https://github.com/samuelflores/molmodel/blob/master/tests/TestProteinDihedrals.cpp)) fail because they hit an assertion deep inside Simbody. The actual backtrace is always the same and looks like this:

    SimTK Exception thrown at ReferencePtr.h:141:
      Error detected by Simbody method ReferencePtr::getRef(): An attempt was made to dereference a null pointer.
      (Required condition 'p!=nullptr' was not met.)

    #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
    #1  0x00007ffff56ae8a4 in __GI_abort () at abort.c:79
    #2  0x00007ffff590c926 in __gnu_cxx::__verbose_terminate_handler() [clone .cold] () from /lib64/libstdc++.so.6
    #3  0x00007ffff59181ac in __cxxabiv1::__terminate(void (*)()) () from /lib64/libstdc++.so.6
    #4  0x00007ffff5918217 in std::terminate() () from /lib64/libstdc++.so.6
    #5  0x00007ffff59184c9 in __cxa_throw () from /lib64/libstdc++.so.6
    #6  0x00007ffff64a85d3 in SimTK::ReferencePtr<SimTK::StateImpl>::getRef (this=0x8129e0)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/./include/SimTKcommon/internal/ReferencePtr.h:140
    #7  0x00007ffff64a8535 in SimTK::ReferencePtr<SimTK::StateImpl>::operator* (this=0x8129e0)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/./include/SimTKcommon/internal/ReferencePtr.h:151
    #8  0x00007ffff649cc1f in SimTK::PerSubsystemInfo::clearAllocationStack<SimTK::CacheEntryInfo> (this=0x8129e0, 
    stack=...) at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/src/State.cpp:187
    #9  0x00007ffff648afe5 in SimTK::PerSubsystemInfo::clearCache (this=0x8129e0)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/src/State.cpp:242
    #10 0x00007ffff648b00c in SimTK::PerSubsystemInfo::clearAllStacks (this=0x8129e0)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/src/State.cpp:245
    #11 0x00007ffff649f9bc in SimTK::PerSubsystemInfo::initialize (this=0x8129e0)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/include/SimTKcommon/internal/StateImpl.h:854
    #12 0x00007ffff648b63d in SimTK::PerSubsystemInfo::restoreToStage (this=0x8129e0, g=...)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/src/State.cpp:338
    #13 0x00007ffff7e719a0 in SimTK::PerSubsystemInfo::invalidateStageJustThisSubsystem (this=0x8129e0, g=...)
    at /home/madcat/Devel/BTU_FGU/MMB_runtime/mmb_deps_debug/simbody/include/simbody/SimTKcommon/internal/StateImpl.h:652
    #14 0x00007ffff648a7e5 in SimTK::StateImpl::operator= (this=0x511a50, src=...)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/src/State.cpp:447
    #15 0x00007ffff648a5ac in SimTK::State::operator= (this=0x4b5020, src=...)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/SimTKcommon/Simulation/src/State.cpp:77
    #16 0x00007ffff758b849 in SimbodyMatterSubsystemRep::convertToQuaternions (this=0x7ac100, inputState=..., 
    outputState=...)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/Simbody/src/SimbodyMatterSubsystemRep.cpp:1797
    #17 0x00007ffff756dbed in SimTK::SimbodyMatterSubsystem::convertToQuaternions (this=0x7ac0e0, inputState=..., 
    outputState=...) at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/Simbody/src/SimbodyMatterSubsystem.cpp:1971
    #18 0x00007ffff73b51e3 in SimTK::ObservedPointFitter::findBestFit (system=..., state=..., bodyIxs=..., 
    stations=..., targetLocations=..., weights=..., tolerance=0.001)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/Simbody/src/ObservedPointFitter.cpp:361
    #19 0x00007ffff73af9b1 in SimTK::ObservedPointFitter::findBestFit (system=..., state=..., bodyIxs=..., 
    stations=..., targetLocations=..., tolerance=0.001)
    at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/simbody/Simbody/src/ObservedPointFitter.cpp:217
    #20 0x00007ffff7c3b631 in SimTK::ObservedPointFitter::findBestFit (system=..., state=..., 
    bodyIxs=std::vector of length 107, capacity 128 = {...}, 
    stations=std::vector of length 107, capacity 128 = {...}, 
    targetLocations=std::vector of length 107, capacity 128 = {...}, tolerance=0.001)
    at /home/madcat/Devel/BTU_FGU/MMB_runtime/mmb_deps_debug/simbody/include/simbody/simbody/internal/ObservedPointFitter.h:71
    #21 0x00007ffff7c29ef3 in SimTK::Compound::fitDefaultConfiguration (this=0x7fffffffd368, 
    atomTargets=std::map with 230 elements = {...}, targetRms=0.0050000000000000001, useObservedPointFitter=true, 
    minimizerTolerance=150) at /home/madcat/Devel/BTU_FGU/MMB/3rdparty/molmodel/src/Compound.cpp:2233

The TestFitMagnesiums code seems to be as close as it gets to a minimal example, although there is still a lot of molmodel code involved. I don't exclude the possibility that this is a molmodel issue because the code hasn't received much maintenance for some time. Is there anything I can do to narrow down the cause of the problem?

Many thanks!

sherm1, 2021-03-26:
Hmmm. That will take some investigation. The presenting problem is that Simbody's State assignment `operator=()` is trying to flush out invalid cache entries in the destination State prior to doing the assignment. (This is occurring due to ObservedPointFitter switching the state back to quaternions from the Euler angles it uses internally, with the assignment at SimbodyMatterSubsystemRep.cpp:1797.) One of the entries in the cache stack turns out unexpectedly to be null. This happens after the optimization has been completed just prior to exiting from `ObservedPointFitter::findBestFit()` (line 361).

This could be a low-level bug in State assignment or initialization, or some higher level heap-trashing issue with ObservedPointFitter or Molmodel. Would be great if you have time to dig in to this further! I could investigate if we have a Simbody-only way to reproduce it, I can't run Molmodel.

MadCatX, 2021-03-26:
I suspected that this was a molmodel issue all along. I took a look at how Simbody uses ObservedPointFitter and eventually fixed the problem with this one-liner:

    - State& state = matchingSystem.updDefaultState();
    + State state = matchingSystem.getDefaultState();

This is the equivalent of `State s = mbs.getDefaultState();` in Simbody's `TestObservedPointFitter` code on L124. Does this make sense or am I just papering over a problem somewhere else?

Thanks a lot for your help with this one.

sherm1, 2021-03-26:
Oh, that makes sense -- writing on the System's internal default state is a bad idea! In fact the [doxygen documentation](https://simbody.github.io/3.7.0/classSimTK_1_1System.html#abe10df3e1df1326164e32098a73772a9) says:
```
/** Don't use this; make a copy of the default state instead and modify
your copy. **/
State&       updDefaultState();
```

I'm not sure _exactly_ why that caused the observed problem, but I suspect this is the right fix. I'll close this one -- please file a new issue if you run into that problem again.

closed 2021-03-26 by sherm1