simbody/simbody #717: Failed to compile (Install) simbody with vs2019

YongchuanYu, 2021-05-28:
When I use VS 2019 to install Simbody in Windows follow the instruction, no matter which type I chosed, "debug", "release" or others, when I right click on "ALL BUILD",  it always has several errors like :“fatal error LNK1104: can not open file '..\..\..\..\..\..\RelWithDebInfo\SimTKmath.lib", 
and also "binary '*': no operator found which takes a left-hand operand of type 'const_Bidlt' (or there is no acceptable conversion)

But with VS 2017 there is no any trouble.

aymanhab, 2022-01-24:
What version of the code base are you trying to compile? We use vs2019 for nightly ci without an issue on latest master branch, log here
https://github.com/simbody/simbody/runs/4698509654?check_suite_focus=true

sherm1, 2022-01-25:
I also recently built Simbody master using VS2019 and VS2022 (locally on my Windows 10 machine) with no trouble. Can you be more specific about the problem? Any possibility you might have some old files around? Or an old version of CMake or old .sln file? I'm using 3.22.1.

sherm1, 2022-01-25:
(I'm building in 64 bit mode)

YongchuanYu, 2022-01-25:
> 
I didn't succeed with VS 2019 until now, I saw some suggestion under this issue yesterday, Let me try those suggestion.
Actually our open library also has many users, as I can not compile with VS 2019, I must answer those question about the faiure on VS 2019 all the time.

YongchuanYu, 2022-01-25:
> 

> Hey YongchuanYu, I am facing the same issue (Working on VS 2019, tried also with VS 2022) What options/ extra tools did you use to get the compilation successfully done with VS 2017? Thank you

I didn't succeed with VS 2019 until now, I saw some suggestion under this issue yesterday, Let me try those suggestion.
Actually our open library also has many users, as I can not compile with VS 2019, I must answer those question about the faiure on VS 2019 all the time.

YongchuanYu, 2022-01-25:
>

MokaMech, 2022-01-25:
Hey Michael and Ayman,

Here is the setup :
I am trying to install Simbody 3.7 to my W10 machine. I'have installed VS
2017 / 2019 / 2022 and VS Build Tools 2022.

I'm following the steps described in the Github Simbody web page.
I have started from scratch, from the sources downloaded on the Github page
and followed the instructions to build it through Cmake 3.22.1 (default
settings, x64 bit version)
Reaching the step where I'm supposed to open the .sln file and built the
ALL_BUILD (RelWitDbg, x64 mode) file into VS 2019, I faced an error "Build
tools v143 required, no such tools found". I managed to get rid of it.
But now I'm facing some other troubles (see enclosed files)

Thank you for your help,

Thomas

Build started...
1>------ Build started: Project: PlatformFiles_subdir, Configuration: RelWithDebInfo x64 ------
2>------ Build started: Project: Test_Regr - TestVectorMath, Configuration: RelWithDebInfo x64 ------
3>------ Build started: Project: Code - SimTKmath, Configuration: RelWithDebInfo x64 ------
2>TestVectorMath.cpp
3>ContactGeometry_TriangleMesh.cpp
2>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm(7140,1): error C2678: binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)
2>        with
2>        [
2>            _BidIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>
2>        ]
2>C:\SimBody_source\simbody-Simbody-3.7\SimTKcommon\BigMatrix\include\SimTKcommon/internal/VectorIterator.h(76,10): message : could be 'ELT &SimTK::VectorIterator<ELT,SimTK::VectorBase<ELT>>::operator *(void)'
2>        with
2>        [
2>            ELT=SimTK::Real
2>        ]
2>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm(7140,1): message : while trying to match the argument list '(const _BidIt)'
2>        with
2>        [
2>            _BidIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>
2>        ]
2>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm(7264): message : see reference to function template instantiation '_BidIt std::_Insertion_sort_unchecked<_RanIt,_Pr>(const _BidIt,const _BidIt,_Pr)' being compiled
2>        with
2>        [
2>            _BidIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>,
2>            _RanIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>,
2>            _Pr=std::less<void>
2>        ]
2>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm(7294): message : see reference to function template instantiation 'void std::_Sort_unchecked<_RanIt,_Fn>(_RanIt,_RanIt,__int64,_Pr)' being compiled
2>        with
2>        [
2>            _RanIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>,
2>            _Fn=std::less<void>,
2>            _Pr=std::less<void>
2>        ]
2>C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm(7299): message : see reference to function template instantiation 'void std::sort<_RanIt,std::less<void>>(const _RanIt,const _RanIt,_Pr)' being compiled
2>        with
2>        [
2>            _RanIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>,
2>            _Pr=std::less<void>
2>        ]
2>C:\SimBody_source\simbody-Simbody-3.7\SimTKcommon\include\SimTKcommon/internal/VectorMath.h(357): message : see reference to function template instantiation 'void std::sort<SimTK::VectorIterator<ELT,SimTK::VectorBase<ELT>>>(const _RanIt,const _RanIt)' being compiled
2>        with
2>        [
2>            ELT=SimTK::Real,
2>            _RanIt=SimTK::VectorIterator<SimTK::Real,SimTK::VectorBase<SimTK::Real>>
2>        ]
2>C:\SimBody_source\simbody-Simbody-3.7\SimTKcommon\tests\TestVectorMath.cpp(331): message : see reference to function template instantiation 'SimTK::VectorBase<ELT> SimTK::sort<ELT>(const SimTK::VectorBase<ELT> &)' being compiled
2>        with
2>        [
2>            ELT=SimTK::Real
[... 654 lines cut ...]
Error	C2100	illegal indirection	Test_Regr - TestVectorMath	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm	7142	
Error	C2678	binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)	Test_Regr - TestVectorMath	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm	7140	
Error	C2678	binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)	Test_Regr - TestVectorMath	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm	7142	
Error	C2678	binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)	Code - SimTKmath	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm	7140	
Error	C2100	illegal indirection	Code - SimTKmath	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.30705\include\algorithm	7140	
Error	C2678	binary '*': no operator found which takes a left-hand operand of type 'const _BidIt' (or there is no acceptable conversion)	Code - SimTKmath	C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.30.3 [... cut ...]
Error	LNK1104	cannot open file '..\..\..\..\..\..\RelWithDebInfo\SimTKmath.lib'	Test_Regr - cpsAdvDiff_bndL	C:\SimBody_source\simbody-Simbody-3.7\build\SimTKmath\Integrators\src\CPodes\sundials\tests\LINK	1	
Error	LNK1104	cannot open file '..\..\..\..\..\..\RelWithDebInfo\SimTKmath.lib'	Test_Regr - cpsRoberts_dns	C:\SimBody_source\simbody-Simbody-3.7\build\SimTKmath\Integrators\src\CPodes\sundials\tests\LINK	1	
Error	LNK1104	cannot open file '..\..\RelWithDebInfo\SimTKmath.lib'	Test_Regr - TestSpline	C:\SimBody_source\simbody-Simbody-3.7\build\SimTKmath\tests\LINK	1

YongchuanYu, 2022-01-25:
> I also recently built Simbody master using VS2019 and VS2022 (locally on my Windows 10 machine) with no trouble. Can you be more specific about the problem? Any possibility you might have some old files around? Or an old version of CMake or old .sln file? I'm using 3.22.1.

This problem happed in last year, since then I tried 3 or 4 times more, then I shift back to VS2017 and ask all our costumers try VS 2017 instead of VS 2019, that is pity :-(. Now my computer is running a simulation, I will ask my new colleague to try again the VS 2019. Thanks for your answer.

YongchuanYu, 2022-01-25:
> What version of the code base are you trying to compile? We use vs2019 for nightly ci without an issue on latest master branch, log here https://github.com/simbody/simbody/runs/4698509654?check_suite_focus=true

Last year when this problem happened, I tried Simbody 3.7 release version, with latest version of CMake, I will try your suggestion to check again. Thanks for your answer.

MokaMech, 2022-01-25:
Sorry I made a mistake in the post above : error files were generated with VS 2022, but I had exactly the same troubles with VS 2019.

MokaMech, 2022-01-25:
Edit : I managed to build the ALL_PROJECT solution thanks to VS 2017. I checked that Cmake was configured with CMAKE_LINKER = C:/path_to_VS2017/bin/.../linker.exe  ( Error LNK1104 form VS)
Did the same for CMAKE_AR and then opened the .sln file with VS2017.

The issue I reported was for Cmake compiler configured for VS 2022, and .sln file opened with VS 2022. Not functionning until now.

sherm1, 2022-01-25:
@MokaMech I see that @aymanhab and I have both been building from the master branch, while you are building from the 3.7 release branch. I think there have been some fixes in the master branch that may explain why we are not having trouble. Is it possible for you to try getting Simbody source from the master branch to see if that fixes your problem? If so I will make a new release that includes those fixes.

YongchuanYu, 2022-02-10:
> @MokaMech I see that @aymanhab and I have both been building from the master branch, while you are building from the 3.7 release branch. I think there have been some fixes in the master branch that [...]
@sherm1 @aymanhab 
I compile Simbody release version with VS 2022 failed, some errors about link problem, while I compile Simbody master version VS 2022, I succeed, do you know what is the problem?

MuhammadUtoro, 2023-07-25:
Hello everyone!
I am facing the same error as YongchuanYu and MokaMech. This error keeps coming up:
**“fatal error LNK1104: can not open file '............\RelWithDebInfo\SimTKmath.lib"**,
Most of the projects' build are failed, such as:
**Done building project "pendEs.vcxproj" -- FAILED.**
and at the end, only 33 succeeded
**========== Build: 33 succeeded, 168 failed, 0 up-to-date, 0 skipped ==========**
When I tried to build the RUN_TESTS, nothing succeeded:
**========== Build: 0 succeeded, 1 failed, 1 up-to-date, 0 skipped ==========**
I have followed all the steps in Github page and also from the OpenSim page (since I want to use Simbody for OpenSim) and I am currently using Windows 10, Visual Studio 17 (2022) and CMake 3.27.0. For the Simbody, I downloaded the latest version 3.7 and I have also tried with the 3.5.3, however they produced exactly the same errors.
I assumed this has to do with the LAPACK and BLAS, but those things are out of my knowledge.
I appreciate your help