simbody/simbody #725: Are the HuntCrossleyForce dissipation coefficients combined correctly?

tgeijten, 2021-11-17:
In HuntCrossleyForce::calcForce(), the dissipation coefficients of two materials are combined as follows:
```
const Real s1 = param2.stiffness/(param1.stiffness+param2.stiffness);
const Real s2 = 1-s1;
const Real c = param1.dissipation*s1 + param2.dissipation*s2; 
```

The dissipation `c` is a weighted sum based on the relative stiffnesses, yet `s1` represents the relative stiffness of `param2` and not `param1`. Shouldn't it be the other way around? This would make more sense:
 
```const Real c = param1.dissipation*s2 + param2.dissipation*s1;```

Here's a link to the relevant code:
https://github.com/simbody/simbody/blob/a8f49c84e98ccf3b7e6f05db55a29520e5f9c176/Simbody/src/HuntCrossleyForce.cpp#L113-L127

sherm1, 2021-11-17:
Hi, Tom. The way to think about this is to imagine one object as very stiff (effectively rigid) and the other very soft. When they interact, the soft one will do almost all the deforming, and hence will contribute almost all of the dissipation. To test the formula, let's say object1 is very soft and object 2 is rigid (that is, param2.stiffness >> param1.stiffness). We would then hope that the composite dissipation parameter c would just be equal to object1's dissipation parameter.

With that set up, lines 115 and 116 make s1=1 and s2=0, giving c=param1.dissipation. That seems right to me. Do you agree?

tgeijten, 2021-11-18:
Your reasoning indeed makes sense. When a stiff and bouncy object falls on a soft inelastic surface, you would want the dissipation to be determined mostly by the soft surface. Thank you for your quick and insightful response!

sherm1, 2021-11-18:
You're very welcome. And thanks for questioning what's there -- I guess we got this one right but that's no guarantee we didn't screw up elsewhere!

closed 2021-11-18 by sherm1