simbody/simbody #728: About 3d constraint library

yumianhuli2, 2022-01-22:
Hello！
What C++ code files of simbody are associated with the dynamics 3d constraint library？
Thank you！

sherm1, 2022-01-26:
In the [Simbody/src](https://github.com/simbody/simbody/tree/master/Simbody/src) directory there is a set of files with names beginning with `Constraint_`, for example:
[Constraint_PointOnPlaneContact.cpp](https://github.com/simbody/simbody/blob/master/Simbody/src/Constraint_PointOnPlaneContact.cpp)

For each of those there is a public header file in [Simbody/include/simbody/internal](https://github.com/simbody/simbody/tree/master/Simbody/include/simbody/internal)

and a private implementation header in Simbody/src named the same but with "Impl" added.

Is that what you are looking for?

yumianhuli2, 2022-01-30:
> In the [Simbody/src](https://github.com/simbody/simbody/tree/master/Simbody/src) directory there is a set of files with names beginning with `Constraint_`, for example: [...]

It looks like it is, but it would be a lot easier to understand if you had a diagram for each constraint. Thank you！！

closed 2022-01-31 by sherm1