simbody/simbody #729: Suggestion: simple algorithm for proper interpolation of quaternions.

fcanderson, 2022-03-09:
@sherm1

I hope you are doing well personally and are not too busy professionally.  I'm writing, not to report a bug, but to see if it makes sense to incorporate a relatively simple algorithm into Simbody related to quaternions.

I continue working to use Blender, an open-source animation suite (www.blender.org), as a means of visualizing Simbody simulations.  A large part of that work has been to develop Python bindings for Simbody.  Most recently, I've written some C++ classes that translate simulation results into animation keyframes. The keyframe classes have turned out nicely, but I'm not sure they would be of interest to the typical Simbody user and so may not be a good candidate for inclusion into Simbody proper.

Anyway, when it came to interpolating quaternions, I came across a very nice, simple algorithm that is almost essential for
interpolating quaternions properly.  The algorithm is a 2-liner and can be implemented as a static method.  In a nutshell, an issue arises because when a Rotation is converted to a Quaternions in Simbody, the Quaternion is returned in canonical form, as should
be the case. However, sometimes a quaternion needs to be expressed in non-canonical form to yield correct interpolations. How does one decide? The algorithm I came across and implemented ensures that a quaternion is in the correct form.  This issue seemed like it might be encountered often enough that it might make sense to incorporate the algorithm into Simbody.  Of course, it's possible something like it is already in Simbody, and I overlooked it.

I wrote some detailed Doxygen comments for the method, which follow below.  If you think this algorithm potentially has a place in Simbody, I am happy to branch and submit a pull request. Unlike happned with ExponentialSpringForce, I wanted to start a discussion before hitting you with something unexpected.

Doxygen comments and static method below here ----------------------------------------------

    /** Given two successive quaternions, Q₁ and Q₂, in a sequence of
    rotations (as might occur when keyframing rotations for a computer
    animation), ensure that Q2 is in the form (canonical or non-canonical)
    that yields interpolations on the the short arc from Q₁ to Q₂. In
    particular, determine if the rotation angle between Q₁ and Q₂ is less
    than or equal to 180° (π radians). If not, make it so by negating Q₂.
    
    #### Background
    For every quaternion, Q = w + xi + yj + zk = ~(w, x, y, z), that generates
    a rotation by an angle α to some point on the unit sphere, there is
    another quaternion, namely -Q, that generates a rotation to that same point
    but by rotating by an angle 360°-α the other way around the unit sphere.
    If |α| ≤ 180°, Q produces the "short-arc" rotation; the short-arc rotation
    is obtained when Q is expressed in canonical form (i.e., when 0 ≤ w ≤ 1).
    If |α| > 180°, Q produces the "long-arc" rotation; the long-arc rotation
    is obtained when Q is expressed in non-canonical form (w < 0). Switching
    between the canonical (short-arc) and non-canonical (long-arc) forms is
    done simply by negating Q.

    The rotation angle α is given by α = 2*cos⁻¹(w). As described above,
    non-negative values of w result in values of α in the range [0° to 180°].
    For example,

            α = 2*cos⁻¹( 1.000) =   0.0°
            α = 2*cos⁻¹( 0.900) =  51.7°
            α = 2*cos⁻¹( 0.100) = 168.5°
            α = 2*cos⁻¹( 0.001) = 179.9°
            α = 2*cos⁻¹( 0.000) = 180.0°

    Negative values of w result in values of α in the range (180° to 360°]:

            α = 2*cos⁻¹(-0.001) = 180.1° = 360° - 179.9°
            α = 2*cos⁻¹(-0.100) = 191.5° = 360° - 168.5°
            α = 2*cos⁻¹(-0.900) = 308.3° = 360° -  51.7°
            α = 2*cos⁻¹(-1.000) = 360.0° = 360° -   0.0°

    Note that negation of a quaternion results in rotation the other way
    around the unit sphere because both w and the rotation axis are negated.

    #### Algorithm Explanation
    When interpolating from Q₁ to Q₂, as is commonly done in computer
    animation, it is critical to take the short arc from Q₁ to Q₂.
    Paradoxically, taking the short arc from Q₁ to Q₂ can sometimes require
    expressing Q₂ in non-canonical (long-arc) form. As an explanation, a
    concrete example is likely most effective.

    Say Q₁ produces a rotation around the unit sphere by α₁ = 176° and Q₂
    produces a rotation that is just a little bit farther, say α₂ = 184°.
    The desired interpolation from Q₁ to Q₂ should produce angles on the
    short arc between α₁ and α₂ (e.g., 176°, 178°, 180°, 182°, 184°).
    Expressing Q₂ in canonical (short-arc) form, however, would have
    α₂ = -176° (assuming the same rotation axis) and interpolation would
    instead yield angles on the long arc between α₁ and α₂
    (e.g., 176°, 88°, 0°, -88°, -176°). The fact that Q₂, in canonical form,
    represents the short arc rotation around the unit sphere (i.e. a rotation
    of -176° instead of 184°) is not helpful in this situation. In order to
    get the short arc from Q₁ to Q₂, Q₂ must be expressed in non-canonical
    (long-arc) form with α₂ = 184°, not α₂ = -176°.

    Algorithmically, a simple dot product of Q₁ and Q₂ (i.e., ~Q₁ * Q₂)
    reveals if Q₂ needs to be expressed in its alternate form to produce
    interpolations along the short arc from Q₁ to Q₂. If the dot product
    is non-negative, |α₂ - α₁| ≤ 180, and Q₂ should be kept in its
    current form. If the dot product is negative, |α₂ - α₁| > 180, and
    Q₂ must be expressed in its alternate form, which is done simply by
    negating Q₂!
    
    This algorithm is presented and discussed by Prof. Ladislav Kavan of the
    University of Utah in a 2014 class lecture for Physics Based Animation:

            "PBA 2014: 3D Rotations and Quaternions"
            https://www.youtube.com/watch?v=IQEysbB0F0Y

    In particular, see the material from 1:00:00 to about 1:14:00. */

    static void ensureQ2YieldsShortArcFromQ1(const Vec4& q1, Vec4& q2) {
        Real q1q2 = ~q1*q2;  // dot product
        if(q1q2<0.0) q2 = -q2;
    }

Doxygen comments and static method above here --------------------------------------------------

Although I went looking for a concise write-up / explanation of this algorithm, I never found one. I found 30 pages on the topic, but not 3 or 4 paragraphs.  So I wrote the above comments pretty much from scratch.

Best,
Clay

ps - In Simbody, I suppose the method should take proper Quaternions as arguments instead of Vec4's. That way they are normalized, etc.

sherm1, 2022-05-05:
Wikipedia also mentions this in its [Slerp ](https://en.wikipedia.org/wiki/Slerp) entry.
My thought would be that Simbody's Quaternion class should have a `q=q1.slerp(f, q2)` function that returns a quaternion q some fraction f of the shortest way between q1 and q2. That seems better to me than exposing the non-canonical quaternion.

fcanderson, 2022-05-06:
Thanks for the Wikipedia link. I should have found that!

Blender has a slerp() method just like the one you propose:  mathutils.Quaternion.slerp(q2, f).  Blender calls that method internally when it interpolates quaternions.  There's an important difference though. Blender's slerp() does not pick the "shortest way between q1 and q2", which is what I was stymied by.  It sometimes generates quaternions that lie on the long arc between q1 and q2.  I therefore took things into my own hands and altered the quaternions I handed to Blender. Hence the origin of the static method that appears in my previous comment.

Even if SimTK::Quaternion_::slerp(f, q2) were available to me, it wouldn't remove the need to call the static method I wrote (or one like it) however.  I don't have a way (that I'm aware of) of asking Blender to use `SimTK::Quaternion::slerp(f, q2)` for the interpolations that Blender performs internally.

Your proposed slerp() sent me back to dig more into Blender's API, thinking Blender MUST have an approach for ensuring short-arc interpolations.  Having missed it the first few times I looked through the API  again, and I think I've identified the approach. The short arc decision is handled in a method intended to be called prior to the slerp() method.  The call is `q2.make_compatible(q1)`, which changes q2 to the form (canonical or non-canonical) that makes the angle between q1 and q2 less than 180°, which is what my static method does.

Since it is so much faster to do in C++, it would still be nice to have the SimTK Slerp method.  However, I would want two methods:

/// Return a quaternion, expressed in canonical form, that is some fraction f along the shortest
/// path between this quaternion and a specified quaternion q2.
```
Quaternion_ Quaternion_::slerp(const RealP& f, const Quaternion_& q2) const {
     // Ensure interpolation with q2 will result in the short arc quaternions
     Quaternion q2short = this->shortArc(q2);
     // Interpolate using *this and q2short
     q = ... {slerp from Wikipedia }
     return q;
}
```

/// Return a quaternion which, when used in slerp, will yield quaternions on the short arc between
/// this quaternion and a specified quaternion, q2. The returned quaternion will be equal to
/// q2 or -q2 and therefore may be in non-canonical form.
```
template <class P> Quaternion_<P>
Quaternion_<P>::shortArc(const Quaternion_<P>& q2) const {
    RealP this_dot_q2 = ~(*this) * q2;
    Vec4P q(q2.asVec4());
    if(this_dot_q2 < 0.0) { q = -q; }
    return Quaternion_<P>(q);
}
```
I'd use the `slerp()` method when resampling quaternions for different video frame rates and playback speeds. As a last step, I'd use the `shortArc()` method to express the quaternions in a form that won't require a call to Blender's q2.make_compatible(q1).  I can certainly make do without a public `shortArc()` method, but it would be nice to have.

hwy1992129, 2022-05-11:
Hello, I am working on OpenSim and Blender. I am interested in your work on Simbody and Blender. Could I learn about what you are doing? Thanks.

fcanderson, 2022-05-11:
@hwy1992129  I'd be happy to discuss my activities relating to OpenSim, Simbody, and Blender. It's probably better if you and I interact via the email address associated with my GitHub repositories:  fcandergit@gmail.com.  Feel free to send me an email, and please tell me more about your interests and activities.