simbody/simbody #733: Exponential Springs in OpenSim

fcanderson, 2022-05-14:
@sherm1 @aymanhab 

This week I began work to bring the `ExponentialSpringForce` contact model into `OpenSim`.  I discovered that all force producing objects in `OpenSim` inherit from `OpenSim::Force`, which in turn has some generic code (written by @peastman  and Ajay) that interacts with `SimTK::Force`.  The issue is that the `SimTK::ExponentialSpringForce` class does not inherit from `SimTK::Force` but rather from `SimTK::ForceSubsystem`.  `SimTK::ExponentialSpringForce`, for example, does not implement a `calcForce()` method.

It seems to me that the easiest path to incorporating `ExponentialSpringForce` into `OpenSim` is to rework `ExponentialSpringForce` so that it inherits from `SimTK::Force` and adds itself to the `GeneralForceSubsystem`.  I'm thinking (hoping) that this will rework will not be such a big job, that it will mostly just require changing some of the `Impl` details.

I will proceed with this rework.  If either of you have any input about this, I am all ears. For instance, there may be a reason, given the features of `ExponentialSpringForce`, that will make inheriting from `SimTK::Force` a difficult path to walk down.

In the past 2 weeks, it is unlikely that someone has started using `SimTK::ExponentialSpringForce`.  If someone has, then there may be some adjustments required when `ExponentialSpringForce` becomes a `SimTK::Force`.  I hoping the adjustments will be minor and a good case in point for hiding the `Impl` details.

-Clay

fcanderson, 2022-05-16:
I successfully reworked the ExponentialSpringForce class to derive from SimTK::Force, instead of being its own ForceSubsystem. 
 All tests are passing.

Once I make a few necessary adjustments to the documentation and do some more testing, I will submit a pull request.

@sherm1, I'm sorry to ask you to review the code once again (after such a thorough go-round the first review).  The changes are almost entirely in ExponentialSpringForceImpl, and they are rather systematic.  One question I do have is how the method SimTK::Force::calcForce() works compared to ::realizedDynamics().  It looks like calcForce() is intended to simply takes its place.

sherm1, 2022-05-16:
Happy to re-review. Just point me to the relevant changes so I don't have to review previously-reviewed stuff.

Yes, IIRC the Force subsystem invokes calcForce() on each Force element from its own realizeDynamics() function.

fcanderson, 2022-05-16:
Thanks Sherm.

Makes sense about calcForce() being called from the Force subsystem's own realizeDynamics() function.

I will leave a comment on the PR suggesting an efficient way to identify the code changes.

fcanderson, 2023-01-17:
Sherm finished the review. The new inheritance from Force works fine.  All tests are passing.  The code was merged in May 2022.  Closing this issue.

closed 2023-01-17 by fcanderson