simbody/simbody #736: Elastic Foundation Force Computation Error

charlesdranoff, 2022-06-16:
I have run into two distinct but related problems when running a forward dynamics simulation using Simbody’s Elastic Foundation Force to model contact between a falling brick (with a contact mesh) and a contact half space:

1. Faces which should not be flagged as “in contact” are producing force
2. The angle of the face is not accounted for in the force computation

This simulation is unidirectional with the y-translation as the only unlocked degree of freedom. The properties of the brick are as follows:

- 0.2 m x 0.2 m x 0.2 m
- Mass = 500 kg
- Rotated 30 degrees around the x-axis
- COM = 2.2 m above the ground
- Simple triangular mesh applied with two triangular meshes per face, totaling 12 triangles
- Markers positioned at the centroid of each triangle (where the spring is placed)

The half space was rotated by -π/2 radians and is coincident with the ground. The model file is provided in the folder. The motion file from the forward dynamics simulation is provided in the folder. A point kinematics analysis is performed on the markers and the net force of contact between the brick and the half space is reported. The analysis file is provided in the folder. 

The net force reported is the sum of all of the force contributions from each triangular mesh in contact with the half space. It appears that the faces with positive (Right 1 & 2 and Top 1 & 2) or no (Front 1 & 2 and Back 1 & 2 ) vertical surface normal components are contributing to the net force although they should not be in contact with the half space even when the brick appears to “pass through” the floor completely. 

I have generated a Python code (provided in the folder) to replicate Simbody’s ElasticFoundationForceImpl::processContact() function, which uses the position and velocity output files (provided in the folder) as well as a user-defined Excel file (provided in the folder) detailing the stiffness, dissipation coefficient, and area of each spring as well as the vector normal to the face of each triangular mesh and the initial height of the spring in the ground frame. When running the code, select all of the position files (12) followed by all of the velocity files (12). The python code closely matches the force table output from Simbody (provided in the folder), validating the python computation. By selecting a time point in which the block appears completely below the half space (e.g. t = 0.71241100 s), we can compute the force contribution from each spring individually. Two problems become apparent:

- For this time point, all faces have a nonzero force contribution, but only “Bottom 1,” “Bottom 2,” “Left 1,” and “Left 2” should contribute as these faces have a negative vertical surface normal component
- It appears that the force registered from each face is not componentized according to the 30-degree offset and instead is reported as the resultant force (normal to the contact half space). 

A proposed solution is to modify the code (provided in the folder) by creating a list (compression_state) formed by the dot product of the unit vectors normal to each mesh with the unit vector normal to the contact half space. The force on each spring is only computed if the compression_state < 0; if compression_state = 0, the contact surfaces are orthogonal (i.e. Front 1 & 2 and Back 1 & 2 ) and if compression_state>0, the block face should not come into contact with the half space (i.e. Right 1 & 2 and Top 1 & 2). This portion of the solution accounts for excluding the force contributions from the appropriate faces. To address the componentization of the force, f (defined in line 175 of the source code) should be multiplied by compression_state instead of forceDir as it is in line 176 in the source code. Since forceDir is a function of displacement and this is a unidirectional test case, the only nonzero forceDir component is in the y-direction, but since the block falls at an angle of 30 degrees rotated around the x-axis, the displacement of the springs on the bottom and left faces would occur in both the y- and z-directions. Thus, multiplying by compression_state should account for the componentization. 
[Bug_Report.zip](https://github.com/simbody/simbody/files/8919645/Bug_Report.zip)

sherm1, 2022-06-16:
Thank you for the detailed report!

Once concern I have is that OpenSim is still using the deprecated ElasticFoundationForce element, rather than the preferred implementation as a ContactForceGenerator (see [CompliantContactSubsystem.cpp:1824](https://github.com/simbody/simbody/blob/master/Simbody/src/CompliantContactSubsystem.cpp#L1824) for the force calculation). Likely that has similar behavior though -- can you tell by looking at the code?

The Elastic Foundation model has many limitations -- ignoring face normals is one, and inability to deal with fully-buried objects is another. Usually the cure is to make a denser mesh, and to avoid simulations in which objects become embedded in one another. Are those feasible in your case?

cc @aymanhab

sherm1, 2022-06-16:
BTW although Simbody's ElasticFoundation models require fairly dense meshes to give reasonable behavior, we have recently developed a new method ("Pressure Field Contact") which is far better and gives smooth forces for any mesh, however coarse. Here is the [published](https://ieeexplore.ieee.org/abstract/document/8968548) and [arXiv](https://arxiv.org/abs/1904.11433) versions of a paper describing it.

The only implementation of that I currently know of is in Drake, but it would be a great addition to Simbody and OpenSim if anyone is interested in tackling that.

charlesdranoff, 2022-06-16:
Thank you for your quick response. I really appreciate you taking the time to help me. I will look into the preferred implementation for the force computation and see if I can refine my mesh to obtain more accurate results. I will keep you updated with any progress. Thank you again for all your help.