simbody/simbody #737: Compile error C2678 when building ALL_BUILD in Visual Studio

volmen3, 2022-06-22:
I encounter some problems when building the ALL_BUILD project in Visual Studio with following specifications:
- Visual Studio 2022 17.2.5 (same problem on VS 2019)
- MSVC v143 - VS 2022 C++-x64/x86-Buildtools
- Building "RelWithDebugInfo" and x64 configuration

I get some compiler error C2678 ([build_log.txt](https://github.com/simbody/simbody/files/8962136/build_log.txt)), complaining about some conversion problems.

sherm1, 2022-06-23:
Can you say which version of Simbody source code you are using? Is this the current master branch from GitHub or an older version? If it's older the problem may have been fixed already in master.

volmen3, 2022-06-23:
I use "Simbody release 3.7", which I've downloaded from here: https://github.com/simbody/simbody/releases

volmen3, 2022-06-24:
Indeed, that was the problem. The compilation and installation process of the master branch (using configuration described above) went well. 

Regarding the release versions: The release version 3.6 has the same problem as 3.7 and using an older compiler version such as MSVC v142 doesn't solve the problem. Just wanted to point out, that it's not only the last release which is affected.

keaidedagui, 2024-01-03:
So in the end, how do you solve this problem? I really want to know the answer.

sherm1, 2024-01-04:
@keaidedagui -- currently the fix is to build from the master branch in GitHub rather than from one of the packaged releases.