simbody/simbody #739: Where is the PIN constraint equation

kakuhin1984, 2022-07-14:
Hello! Dear SimBody Team!
 where is the constraint equation for MobilizedBody::Pin in the constraint classes source file?

closed 2022-07-14 by sherm1

sherm1, 2022-07-15:
Yes, we prefer mobilizers (internal coordinate joints) since they are much more efficient and not subject to drift. Those are limited to forming a tree topology though. Simbody also supports arbitrary constraints, and those can also be used to create joints. However, if you have a multibody system with a loop topology, we recommend that you simply cut one of the bodies to form a tree, then glue it back together with a Weld constraint. That way all your joints are modeled consistently as mobilizers, and the one constraint (6 constraint equations) is a very simple one.

yumianhuli2, 2022-07-16:
> Yes, we prefer mobilizers (internal coordinate joints) since they are much more efficient and not subject to drift. Those are limited to forming a tree topology though. Simbody also supports [...]

ok,thank you!