simbody/simbody #747: The OBJ file parser should ignore all kinds of whitespace for face (`f`) commands

adamkewley, 2022-11-03:
Minor bug, but it popped up when a user supplied custom-computed mesh data to OpenSim Creator (see: https://github.com/ComputationalBiomechanicsLab/opensim-creator/issues/473)

The bug is that SimTK's OBJ parser will fail to parse OBJ files that use tabs as delimiters in the face (`f`) command.

A quick search on the internet indicates that OBJ fields do not have a strictly specified field delimiter. Examples and files I have found on the internet indicate that fields may be separated with:

- A single space
- A newline character (with `\` as a control character)
- A single tab
- A single "other" whitespace character (vertical tabs)
- (maybe) Multiple instances of the above

E.g. Blender's OBJ parser handles all of the above by just treating all control characters the same way:

- (call point): https://github.com/blender/blender/blob/a16dd407b3d3f830c40646fcda5ea5b89b3ad4ee/source/blender/io/wavefront_obj/importer/obj_import_file_reader.cc#L499
- (implementation): https://github.com/blender/blender/blob/a16dd407b3d3f830c40646fcda5ea5b89b3ad4ee/source/blender/io/wavefront_obj/importer/obj_import_file_reader.cc#L390

SimTK's OBJ loader currently handles this fairly liberal specification for vertex commands (`v`), but not face commands (`f`) in OBJ files. This is because the vertex parser uses `std::istream::operator>>` to read each vertex element:

- https://github.com/simbody/simbody/blob/d685ed2/SimTKcommon/Geometry/src/PolygonalMesh.cpp#L174

Where C++ specifies that the operator:

> Extracts a floating-point value potentially skipping preceding whitespace. (see: https://en.cppreference.com/w/cpp/io/manip/skipws)

But the face commands (`f`) use `std::istream::ignore` with a space character (`' '`):

- https://github.com/simbody/simbody/blob/d685ed2/SimTKcommon/Geometry/src/PolygonalMesh.cpp#L191

Where C++ specifies that `ignore` skips that character multiple times until it doesn't see it (in this bug, until it sees `\t` and stops).

The solution to this is to change that code section to skip as much *whitespace* as possible, rather than specifically the space character.

adamkewley, 2022-11-03:
I might be able to fix this myself and PR it - I'm just writing it up so that it's clear what the issue is (I'll add some basic repro files etc. later)

tandede, 2026-08-25:
I’d like to work on this if it is still available. I plan to replace the face parser’s literal-space handling with whitespace-aware tokenization, then add focused OBJ regression coverage for tabs and mixed whitespace while preserving slash-delimited face indices. I’ll keep the change scoped to #747.

closed 2026-08-25 by sherm1