simbody/simbody #751: TestBigMatrix fails on mips64el and ppc64el architectures

merkys, 2022-12-01:
Simbody 3.7 fails TestBigMatrix when built on Debian on mips64el and ppc64el architectures:
```
test 12
        Start  12: TestBigMatrix

12: Test command: obj-mips64el-linux-gnuabi64/TestBigMatrix
12: Working Directory: obj-mips64el-linux-gnuabi64/SimTKcommon/tests
12: Test timeout computed to be: 1500
12: f(v)=~[1,2,3]
12: exception: SimTK Exception thrown at TestBigMatrix.cpp:100:
12:   Internal bug detected: Test condition failed.
12:   (Assertion '(X*vs == -(X*-vs))' failed).
12:   Please file an Issue at https://github.com/simbody/simbody/issues.
12:   Include the above information and anything else needed to reproduce the problem.
  9/108 Test  #12: TestBigMatrix ..................................***Failed    0.02 sec
f(v)=~[1,2,3]
exception: SimTK Exception thrown at TestBigMatrix.cpp:100:
  Internal bug detected: Test condition failed.
  (Assertion '(X*vs == -(X*-vs))' failed).
  Please file an Issue at https://github.com/simbody/simbody/issues.
  Include the above information and anything else needed to reproduce the problem.

```
Full build logs: [mips64el](https://buildd.debian.org/status/fetch.php?pkg=simbody&arch=mips64el&ver=3.7%2Bdfsg-2&stamp=1669708062&raw=0) and [ppc64el](https://buildd.debian.org/status/fetch.php?pkg=simbody&arch=ppc64el&ver=3.7%2Bdfsg-2&stamp=1669706492&raw=0). Log files are ~100 MB in size.

sherm1, 2022-12-01:
Is this something that was working previously and stopped working? If so, any idea what broke it? (Simbody code change, build compiler change, etc?)

merkys, 2022-12-02:
Yes, TestBigMatrix used to pass on mips64el and ppc64el architectures. Doing some log bisecting I find the following:

| Simbody version | C/CXX compiler version | TestBigMatrix |
| --- | --- | --- |
| 3.6.1 | 8.3.0 | Passed |
| 3.6.1 | 12.2.0 | Failed |

Thus it seems that the problem for these architectures is the update of C/CXX compiler from 8.3.0 to 12.2.0.

sherm1, 2022-12-04:
I'm not sure what to do with this. The problem appears to be an optimizer bug in the mips and ppc backends for the newer compiler, but I don't have access to those so can't confirm that compiling with optimization off allows the test to pass. Do you have an idea for how to address this problem?

merkys, 2022-12-05:
I can try disabling optimization for these architectures just to see if that helps. How can I do that?

sherm1, 2022-12-31:
@merkys -- sorry for the delayed response!

As an experiment, please try modifying line [CMakeLists.txt:400](https://github.com/simbody/simbody/blob/master/CMakeLists.txt#L400) at the top level to read
`set(GCC_OPT_DISABLE -O1)` or `-O0` if it still fails at O1.
Then regenerate with CMake and rebuild to see of all tests pass.

If that works we can PR a conditional on the architecture.

merkys, 2023-01-03:
Just checked, neither `-O1`, nor `-O0` makes TestBigMatrix pass, alas.

sherm1, 2023-01-03:
Rats! That's surprising. Thanks for trying. Is the compiler you're using gcc or clang?